horizon-eda/horizon #468: Bug report: Assertion failed during pool update

jbayless, 2020-07-20:
I'm not sure exactly what steps reproduce this error. I was working on creating a new package for the C2512 surface-mount passive component footprint. I'd created the new package and saved it in its own folder alongside the other smd passives, but then noticed that it didn't show up when I searched for it in the pool. So I clicked "update pool" and, during the update, hit this assertion error.
 
[image]

Windows 10. Horizon v1.1.1 Blue sky, commit 63ade81b

Is there a log somewhere that I can upload for you?

PS: This error is only a minor disruption for me. I'm just reporting it because I assume you'd want to know.

jbayless, 2020-07-20:
Update: It happens every time I click 'Update Pool'.

When I deleted `packages/passive/smd/C2512/package.json`, the problem went away.

Here are the file contents:

```
{
    "holes": {},
    "name": "Pad",
    "padstack_type": "top",
    "parameter_program": "",
    "parameter_set": {},
    "parameters_required": [],
    "polygons": {},
    "shapes": {},
    "type": "padstack",
    "uuid": "6c86cc38-d16c-41ef-b8b0-9496c2b18519",
    "well_known_name": ""
}
```

Looks like I accidentally created a padstack instead of a package. My bad! But it might still be a good idea to gracefully handle this exception.

carrotIndustries, 2020-07-20:
Rather than the ugly failed assertion we now get a nicer exception:

[image]

closed 2020-07-20 by carrotIndustries

jbayless, 2020-07-20:
Nice work. You're really fast!

commit 55505e6c (2020-07-20): use json::at rather than [] to avoid crashes

fixes #468