horizon-eda/horizon #469: Text tool: Issues with inserting newlines

jbayless, 2020-07-20:
There appear to be some issues with inserting newlines in the text tool.

Here's a sequence of screenshots that shows the experience right now:

[image]

Observed behaviour:

- In the schematic editor, when I create a new text object, there's no option to add newlines. Shift+Enter does not appear to have an effect.
- However, if I go back and edit the object _after_ placing it, the hint at the bottom of the text editor appears, and I can add a single newline via Shift+Enter.
- But after adding one newline, the hint at the bottom changes to "Press Shift+Enter to finish editing". So I have to add additional newlines by hitting Enter without shift. This is unexpected!

Desired behaviour:
- It should be possible to add newlines when the text is being created, not only when editing already-placed text.
- I know that many programs use Shift+Enter to add newlines and Enter to submit a form, and many other programs use Enter to add newlines and Shift+Enter to submit a form. But Horizon is the first I've encountered that uses the first method for the first newline, and then the second method for subsequent newlines! I think it would be better to pick one method and be consistent.

Windows 10. Horizon v1.1.1 Blue sky, commit 63ade81

carrotIndustries, 2020-07-20:
Well, the first one was easy: bebd57a8bfc9570931c66a602ae5494d9d853d69

The second one is intentional and at least one other user thinks this is the right way. Yes, it might be a bit confusing at first, but I believe it's the right thing to do:

When editing a single line of text, it's muscle memory to press Enter and it's consistent with the single line editor. Shift+Enter then is more to be seen as a hotkey to get into the multi-line editor.

When editing multiple lines of text, Enter should only insert a newline as I find it irritating when Enter in text areas does the 'okay' action.

jbayless, 2020-07-20:
Sounds good! The first problem was the more important one. If the second behaviour is intentional and you like it that way, then I don't have a problem with it.

closed 2020-07-20 by jbayless