horizon-eda/horizon #470: Feature request: part generator: ability to suppress pads

bob-u, 2020-07-22:
Some IC do not use all of the pads in given standard package.  It would be nice to be able to suppress generation of certain pads in the standard package so pad numbers come up correctly and there is no need to manually delete them.

[image]

[image]

carrotIndustries, 2020-07-23:
> and/or have ability to renumber selected pads 

Select the pads you want to renumber and use the "renumber pads" tool.

>  so pad numbers come up correctly

The pad numbers in your screenshot are most likely correct. All packages I've seen that are missing pins from a standard package usually keep its pin numbering.

closed 2020-07-23 by carrotIndustries

bob-u, 2020-07-23:
As you can see on dimensional drawing for the part - pads are actually numbered as they are.  The IC is TMC5160.  

Renumber pads tool is great.  Thanks.