horizon-eda/horizon #472: Issue: not all pads listed

bob-u, 2020-07-22:
Select Padstack window does not list all of the pads.  In example below I want to add EP padstack, which is available in the pool (right) yet not listed as a choice:

[image]

carrotIndustries, 2020-07-22:
This is working as intended. Padstacks come in two flavors:

 - Global padstacks: available to all packages
 - Package-local padstacks: only available to the package they belong to

The exposed pads are of the latter variety.

closed 2020-07-22 by carrotIndustries

bob-u, 2020-07-22:
I though that would be the case so I made a duplicate - but it copies the package it belongs to as well.  How can I make a duplicate to free it from the part?

carrotIndustries, 2020-07-22:
> How can I make a duplicate and free it from the part?

Duplicate the padstack, not the package. There are not parts involved here: https://horizon-eda.readthedocs.io/en/latest/pool-why.html#

bob-u, 2020-07-23:
Sorry, I was not clear.  I duplicated the padstack (not package), however, duplicate padstack was still not showing up in the list of of padstacks to choose from.  Presumably, as you explained, because it was still being treated as package-local padstack, not global.

carrotIndustries, 2020-07-23:
If you put the padstack in a subdirectory of `padstacks`, it's global by default.

Anyhow, the pool didn't get updated automatically, after duplicating some items, this might also have contributed to the padstack not showing up without a manual pool update or saving the padstack. 8a978bf4a0752dedb13e1365cf3f0d37678c96c6 fixes this.