horizon-eda/horizon #474: Feature request: ability to apply pad properties in bulk

bob-u, 2020-07-22:
Feature request: ability to apply pad properties to selected pads.  Currently properties can only be applied one pad at a time:

[image]

[image]

rroohhh, 2020-07-22:
If you press the checkmark they get applied to every pad.

bob-u, 2020-07-22:
Lol, it did not occur to me that this button was for that ( I thought it was to apply the individual parameter).  I would suggest to make this button 'All' to follow the convention, as in the right hand menu.

Speaking of apply - it would be nice to have an apply button (or changes made in real-time) so one can see the changes  them w/o exiting the window.  Especially since it's possible to select individual pads with the selector within the window.

atoav, 2020-07-27:
> Speaking of apply - it would be nice to have an apply button (or changes made in real-time) so one can see the changes them w/o exiting the window. Especially since it's possible to select individual pads with the selector within the window.

If there are performance worries this could be a checkbox like in Photoshop filters: `[ ] Preview`

As for the apply to all button, I could imagine `All` or a double checkmark `✓✓` (maybe a bit closer together in terms of kerning, but there is no UTF-8 Symbol for that)

commit e2bf3b2d (2020-12-29): tool edit pad: make dialog non modal

closes #474
closed 2020-12-29 by carrotIndustries