horizon-eda/horizon #476: SEGV cloning community pool repo

Eric678, 2020-07-23:
On fedora 31 built current master without issue.  
First run does not get past startup - accepting default repo & dest dir after selecting download community pool from welcome screen:

`Thread 20 "horizon-eda" received signal SIGSEGV, Segmentation fault.`
`[Switching to Thread 0x7fffd5ffb700 (LWP 81472)]`
`0x00007ffff63cc7fd in on_header_field (parser=0x7fffc00fc8a0, `
`    str=0x7fffc00fc979 "Server: GitHub Babel 2.0\r\nContent-Type: application/x-git-upload-pack-advertisement\r\nTransfer-Encoding: chunked\r\nExpires: Fri, 01 Jan 1980 00:00:00 GMT\r\nPragma: no-cache\r\nCache-Control: no-cache, max-"..., len=6)`
`    at ../src/transports/http.c:296`
`296            http_subtransport *t = ctx->t;`
`(gdb) bt`
`#0  0x00007ffff63cc7fd in on_header_field (parser=0x7fffc00fc8a0, `
`    str=0x7fffc00fc979 "Server: GitHub Babel 2.0\r\nContent-Type: application/x-git-upload-pack-advertisement\r\nTransfer-Encoding: chunked\r\nExpires: Fri, 01 Jan 1980 00:00:00 GMT\r\nPragma: no-cache\r\nCache-Control: no-cache, max-"..., len=6)`
`    at ../src/transports/http.c:296`
`#1  0x00007ffff51d71a2 in http_parser_execute (parser=parser@entry=0x7fffc00fc8a0, settings=settings@entry=0x7fffc00fc8c0, `
`    data=<optimized out>, len=<optimized out>) at ../http_parser.c:1380`
`#2  0x00007ffff63cd924 in http_stream_read (stream=0x7fffc01298c0, buffer=0x7fffc00e55d0 "", buf_size=65536, `
`    bytes_read=0x7fffd5ff8af0) at ../src/transports/http.c:742`
`#3  0x00007ffff63cf0f2 in git_smart__recv_cb (buf=0x7fffc00e55a8) at ../src/transports/smart.c:25`
`#4  0x00007ffff63d1120 in git_smart__store_refs (t=t@entry=0x7fffc00e5410, flushes=2) at ../src/transports/smart_protocol.c:55`
`#5  0x00007ffff63cf82e in git_smart__connect (transport=0x7fffc00e5410, `
`    url=0x7fffc00d7320 "https://github.com/horizon-eda/horizon-pool.git", cred_acquire_cb=0x0, cred_acquire_payload=0x190bcc0, `
`    proxy=0x7fffd5ff8e00, direction=0, flags=0) at ../src/transports/smart.c:253`
`#6  0x00007ffff63afae0 in git_remote_connect (remote=remote@entry=0x7fffc00d7aa0, `
`    direction=direction@entry=GIT_DIRECTION_FETCH, callbacks=callbacks@entry=0x7fffd5ff8d88, proxy=0x7fffd5ff8e00, `
`    custom_headers=<optimized out>) at ../src/remote.c:648`
`#7  0x00007ffff63b13b1 in git_remote_fetch (remote=0x7fffc00d7aa0, refspecs=refspecs@entry=0x0, `
`    opts=opts@entry=0x7fffd5ff8d80, `
`    reflog_message=0x7fffc00e2310 "clone: from https://github.com/horizon-eda/horizon-pool.git") at ../src/remote.c:966`
`#8  0x00007ffff6352797 in clone_into (branch=0x0, co_opts=0x7fffd5ff8e48, opts=0x7fffd5ff8ed8, _remote=0x7fffc00d5f10, `
`    repo=0x7fffc00e1e80) at ../src/buffer.h:139`
`#9  git_clone (out=out@entry=0x7fffd5ffa2a0, url=<optimized out>, local_path=0x7fffc00dfc70 "/home/user/horizon-pool/.remote", `
`    _options=_options@entry=0x7fffd5ffa360) at ../src/clone.c:433`
`#10 0x0000000000699ff6 in horizon::PoolProjectManagerAppWindow::download_thread (this=0x190bcc0, gh_username=..., gh_repo=..., `
`    dest_dir=...) at /usr/include/c++/9/bits/basic_string.h:2300`
`#11 0x00000000006992ad in std::__invoke_impl<void, void (horizon::PoolProjectManagerAppWindow::*)(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >), horizon::PoolProjectManagerAppWindow*, Glib::ustring, Glib::ustring, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > (`
`    __t=@0x1ab35b8: 0x190bcc0, __f=`
`    @0x1ab35c0: (void (horizon::PoolProjectManagerAppWindow::*)(class horizon::PoolProjectManagerAppWindow * const, class std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, class std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, class std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)) 0x6999b0 <horizon::PoolProjectManagerAppWindow::download_thread(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)>) at /usr/include/c++/9/bits/char_traits.h:300`
`#12 std::__invoke<void (horizon::PoolProjectManagerAppWindow::*)(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >), horizon::PoolProjectManagerAppWindow*, Glib::ustring, Glib::ustring, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > (__fn=`
`    @0x1ab35c0: (void (horizon::PoolProjectManagerAppWindow::*)(class horizon::PoolProjectManagerAppWindow * const, class std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, class std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, class std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)) 0x6999b0 <horizon::PoolProjectManagerAppWindow::download_thread(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)>) at /usr/include/c++/9/bits/invoke.h:95`
`#13 std::thread::_Invoker<std::tuple<void (horizon::PoolProjectManagerAppWindow::*)(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >), horizon::PoolProjectManagerAppWindow*, Glib::ustring, Glib::ustring, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >::_M_invoke<0ul, 1ul, 2ul, 3ul, 4ul> (this=0x1ab3558) at /usr/include/c++/9/thread:244`
`#14 std::thread::_Invoker<std::tuple<void (horizon::PoolProjectManagerAppWindow::*)(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >), horizon::PoolProjectManagerAppWindow*, Glib::ustring, Glib::ustring, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >::operator() (this=0x1ab3558)`
`    at /usr/include/c++/9/thread:251`
`#15 std::thread::_State_impl<std::thread::_Invoker<std::tuple<void (horizon::PoolProjectManagerAppWindow::*)(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >), horizon::PoolProjectManagerAppWindow*, Glib::ustring, Glib::ustring, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > >::_M_run (this=0x1ab3550) at /usr/include/c++/9/thread:195`
`#16 0x00007ffff62053d4 in std::execute_native_thread_routine (__p=0x1ab3550)`
`    at ../../../../../libstdc++-v3/src/c++11/thread.cc:80`
`#17 0x00007ffff7f924e2 in start_thread () from /lib64/libpthread.so.0`
`#18 0x00007ffff5f046a3 in clone () from /lib64/libc.so.6`

What might I be missing?
Thanks

carrotIndustries, 2020-07-23:
Googling `on_header_field libgit2` lead me to https://bugzilla.redhat.com/show_bug.cgi?id=1814897
Since your observations match the one in the bugtracker, I'm closing this issue as it's 99% not a bug in Horizon EDA.

closed 2020-07-23 by carrotIndustries

Eric678, 2020-07-23:
Thanks Lucas, I did switch to f32 and the fault disappeared, so this is correct.