horizon-eda/horizon #477: Pool manager should automatically preview the first entry

Vorlent, 2020-07-24:
This is not really a bug. Just a minor UX annoyance. For a first time user like me it's pretty confusing what the purpose of the blue area was until I selected something. If the preview showed the first part as soon you open the pool manager it would become pretty obvious why all that space is taken up. It would also look much nicer than a boring blue background. Oh and the cursor doesn't work properly if you haven't made a selection yet. It's stuck in the upper left corner.

[image]

If this change is okay I could probably write the pull request myself. It doesn't really look like a huge code change to me.

bob-u, 2020-07-24:
As a user, I agree with you that empty blue screen can be confusing.  Defaulting to first entry would be a quick fix.

I think most programs, if nothing is selected, would show text or graphic describing what this area is about:

<name of selected tab>
        Preview Area

More ambitious screen would add graphic representation of what the tab is about so user gets a visual hint on what this is about

I think doing the above would be more consistent, might be more helpful (especially with the visual hint) and not dependent on what is in the pool.  Might be boring to see for seasoned user however :)

atoav, 2020-07-27:
>  on second thought, I think idea of just selecting the first part in the list would be enough (as log as there is anything in the pool).

I like that better as well