horizon-eda/horizon #478: Bug report: Exception when changing symbol unit

jbayless, 2020-07-25:
I got the following error message:

```
Exception in signal handler

std::exception: map::at
Looks like this is a bug. Please report this and describe what you did previously.
```

I'll do my best to describe the circumstances. This happened while I was creating a set of symbols, units, and entities for various types of comparators. Because of the way the "Duplicate Unit" menu works, I haven't found an efficient way of creating a new unit that is similar to an existing one (for example, an open-collector output comparator vs a push-pull output comparator). So this was my sequence of actions as best I remember:

1. Duplicated the Opamp symbol in the horizon-pool library. (All of this is being done in the context of my local horizon-pool copy) and renamed it Comparator-OC.
2. Created a Comparator-OC unit. Defined its pins (IN+ / input, IN- / input, OUT / open-collector).
3. Edited the Comparator-OC symbol to update its Unit to reference the Comparator-OC unit, instead of the Opamp. This wiped away the old opamp pins so I placed the comparator pins.
4. Duplicated the Comparator-OC symbol, and renamed the duplicate Comparator-pushpull, still referencing the Comparator-OC unit.
5. Created a Comparator-PushPull unit. Defined its pins (IN+ / input, IN- / input, OUT / output).
6. Edited the Comparator-PushPull symbol. Updated its unit to reference the Comparator-pushpull unit...
7. At this point I encountered the exception.

Did I do something wrong here? Maybe I have the wrong grasp of the relationship between entities, symbols, and units?

jbayless, 2020-07-25:
Happened again when I created a new unit for a new comparator unit. This one had a built-in voltage reference and an enable signal, so I renamed the pins. Only the 'OUT' pin shared a name with the old set of pins.

When I went to update the symbol I'd created (duplicated from Comparator-OC) and changed its reference to point to this new comparator, this time it deleted the 2 input pins but the 'OUT' pin was still there. When I clicked on the 'OUT' pin to delete it, I encountered this same exception again.

jbayless, 2020-07-25:
1. Update the symbol's unit reference...

[image]

2. One of the pins, which shares its name with the old pins, still remains on the symbol. The old pins have been deleted.

[image]

3. Try to select that pin. This error message pops up when you do.

[image]

carrotIndustries, 2020-07-25:
> Because of the way the "Duplicate Unit" menu works, I haven't found an efficient way of creating a new unit that is similar to an existing one

What's wrong with the duplicate unit dialog?

commit 494f19ae (2020-07-25): fix symbol pin uuid when changing unit

fixes #478
closed 2020-07-25 by carrotIndustries