horizon-eda/horizon #479: Feature request: pan key

alexforencich, 2020-07-25:
Middle click to pan is nice, but it's hard to use on some machines.  For instance, middle click can be shared with the scroll wheel, so it can be hard to middle click and pan without also scrolling (and, as a result, zooming).  It can also be difficult to middle click, not to mention middle click drag, on laptop touchpads.  In other software packages, using spacebar or some other modifier key, possibly in addition to left clicking, can be used to pan.

bob-u, 2020-07-25:
Shift+LMB activates pan also.

alexforencich, 2020-07-25:
So it does.  Is that documented anywhere?

bob-u, 2020-07-25:
Not sure - I saw it in the source code.  SHIFT to pan is a little buggy as it activates a tool (via drag, I believe) when panning when object is selected.

alexforencich, 2020-07-25:
Yeah, shift to pan doesn't work while dragging some object.  Middle click does, but that's a bit of a pain as it's both left + middle click at the same time.  There are definitely problems with doing that on a laptop trackpad, as middle click is commonly indicated by pressing both the left and the right buttons at the same time.

carrotIndustries, 2020-07-25:
397757d240d830d88292c326a07d47943143482d prevents shift-click activating pan.

For laptops, there are other ways to pan:
 - Trackpoint pans by default (hold ctrl to zoom)
 - Scrolling on the touchpad can be configured to pan (works best with two-finger scroll)
 - Touchscreen

As for dragging, you can free up one mouse button by pressing <kbd>m</kbd> to invoke the move tool.

> EDIT: personally I could use a pan key that does not require pressing any mouse buttons to avoid strain.

I've though of this as well, but this didn't work out as the canvas needs to receive motion events even if the cursor left it. If not holding down a mouse button, the canvas only receives events while the cursor is over it.

> So it does. Is that documented anywhere?

Nope, feel free to open a PR in https://github.com/horizon-eda/horizon-docs. It's not documented yet as I thought that new users would randomly try holding down modifiers to see if anything interesting happens. (at least that's what I do)

bob-u, 2020-07-25:
Although I do have a laptop, I use a trackball.  Just to see, I  tried trackpoint zoom and it's buttery smooth.  I thought scroll to zoom was smooth but trackpoint zoom, and presumably touchpad zoom, is on another level.

atoav, 2020-07-27:
There are rare cases when one is totally without a pointing device, and you would be quite lost to create a schematic this way. However there might be cases where having panning on the keyboard might actually be a nice thing..? 

I am not sure if this is worth the effort, but let me propose something:

Right now we have these actions on the Arrow Keys:
- <kbd>←/↑/↓/→</kbd>: Move the selected object by one grid unit
- <kbd>Alt</kbd>+<kbd>←/↑/↓/→</kbd>: Move the selected object by one finer grid unit

Maybe adding something like this would be cool:
- <kbd>Ctrl</kbd>+<kbd>←/↑/↓/→</kbd>: Move the view by n grid units
- <kbd>Ctrl</kbd>+<kbd>Alt</kbd>+<kbd>←/↑/↓/→</kbd>: Move the view by n grid units (finer than the above)

This is quite consistent with the rest of the behavior, still leaves us with <kbd>Shift</kbd>+<kbd>←/↑/↓/→</kbd> for whatever else and might actually be more comfortable than grabbing the mouse/trackpad in some situations (as it allows a stepped precise movement). Also: maybe people could create a vim keybinding for horizon : D

closed 2020-08-18 by carrotIndustries

carrotIndustries, 2020-08-18:
These are the default pan/zoom keybindings:

|Action | Key sequence | Comments|
|-|-|-|
|Pan up|`Ctrl+Up` | |
|Pan up|`KP_Up` | |
|Pan down|`Ctrl+Down` | |
|Pan down|`KP_Down` | |
|Pan left|`Ctrl+Left` | |
|Pan left|`KP_Left` | |
|Pan right|`Ctrl+Right` | |
|Pan right|`KP_Right` | |
|Zoom in|`KP_Add` | |
|Zoom in|`Ctrl+equal` | |
|Zoom out|`KP_Subtract` | |
|Zoom out|`Ctrl+minus` | |

alexforencich, 2020-08-18:
This is good, but wasn't quite what I had in mind.  I'm thinking somewhere closer to middle click emulation, but using a key on the keyboard instead of a mouse button, or in combination with left click.  If it's separate from left click, then you can use it to pan while dragging, for example.  Shift to pan was nice, but it seems like that was accidental and was removed?  In which case, something more permanent would be worth considering.

carrotIndustries, 2020-08-18:
> Shift to pan was nice, but it seems like that was accidental and was removed?

No, shift-left click is an actual feature. As for dragging, all actions that require dragging can be performed differently without holding down the left button, such as pressing m for move rather than dragging.

alexforencich, 2020-08-18:
Ah I see, the way you phrased that earlier seemed like shift-click wasn't supposed to activate pan.

commit 1d57d9c6 (2020-08-18): add actions for pan and zoom

fixes #479