horizon-eda/horizon #480: Feature request: Zoom and center

bob-u, 2020-07-27:
Feature Request: When zooming in, center portion of the screen under cursor when scroll event started.  This feature would be useful when reviewing schematic or board or just to avoid panning.  It seems that all the functions are in Horizon already so I tried to hack an example:

[image]

[image]

atoav, 2020-07-27:
I think that would be a great option to have. I am used to both scrolling behaviors from different applications, but in this case I'd probably like something like this more.

Maybe something like this would work:
- there is a _Default Zoom_ and an _Alternative Zoom_
- _Default Zoom_ is just <kbd>Scroll</kbd>
- _Alternative Zoom_ is <kbd>Ctrl</kbd>+<kbd>Scroll</kbd>
- Add an Option in `Preferences > Editor > Zoom & Pan` to invert the behavior

This way both things would be fast at hand if needed, but one could still decide on a favored one – what do you thing @carrotIndustries ?

carrotIndustries, 2020-07-27:
To me at least anything that doesn't keep the point under the the cursor where it is is highly irritating.

> It seems that all the functions are in Horizon already so I tried to hack an example ('Zoom in' centers, 'zoom out' works like it is now):

Do you have fork where I could try this?

BTW: Your screenshots are a good example why distros shouldn't use theming for branding: https://blogs.gnome.org/tbernard/2018/10/15/restyling-apps-at-scale/ as the osd-style widgets (hud, action buttons) should have a dark background.

bob-u, 2020-07-27:
> Do you have fork where I could try this?

It's on a different computer right now...

It's basically remembers position of the pointer when zoom_animator starts (and ignores new positions from gtk for the time of animation), calculates xy distance from that point to center of the screen and divides it into equal amount of steps depending on zoom_animator frames in the zoom animation from scale origin to scale target.  Then adjusts offset.x and offset.y by adding computed xy step each frame.

> I think that would be a great option to have. I am used to both scrolling behaviors from different applications, but in this case I'd probably like something like this more.

This is basically a better on eyes (animation preserves continuity) KiCad's center and zoom feature.  I dislike that one because it jumps to center in one frame causing you to loose point of reference.  I did not use that feature in KiCad because of that sudden jump.  Given that Horizon is so much smoother with zoom and pan, I think it would be quite usable.

> To me at least anything that doesn't keep the point under the the cursor where it is is highly irritating.

Everyone works little differently.  I dislike that I have to pan *every time* I zoom in on items around the corners of the screen.  This is one way to address that it and seems to work pretty well.

I have few other ideas that address that:  For example, pan x amount of pixels towards the center when zoom in occurs near a border of the window.  Another, to pan towards center (instead of zooming) with scroll wheel when a special key is pressed - though when reviewing I rather use mouse gestures only, w/o touching keyboard.