horizon-eda/horizon #481: Feedback for Parts Window

atoav, 2020-07-27:
This is feedback on the parts window implemented in the branch [part-win](https://github.com/horizon-eda/horizon/tree/parts-win).

[image]

This is a very welcome addition. I was thinking about such a thing for quite a while, but never dared to ask. Instead of directly giving feedback on the solution @carrotIndustries (thankfully!) created, I will try to ask myself what problems a window like this is meant to solve/help solving, and how it might be used, and then we can go from there and wonder in which direction the whole thing should move.

## Analysis

This is essentially the accessible parts list within horizon. As for now one could export a BOM or use <kbd>Ctrl</kbd>+<kbd>F</kbd> to get a feeling what is on a board/in a schematic. What is such a list used for?
- Definitive list of parts: when you left a power unit of a opamp on the wrong location that could be hard to detect, similarily if you accidentally used two different series of a 100nF bypass cap or something of those sorts: hard to spot on board/schematic, easy to spot in a list
- Searching, finding and filtering parts
- As a tick-off list for placing parts
- Maybe for crude back of the envelope price calculations (although one would probably just use the CSV export instead)
- This could be a comfortable place to shadow/fix/do-not-place/omit-silkscreen/etc filtered selections of parts or allow to edit things like reference designators etc

## On the current solution

The way it has been proposed is very much geared towards parts placement. This is not a bad thing – as mentioned above I was dreaming about such a thing myself. But as I thought a little more about it I was thinking it might be more wise to make this a general "Parts List" window, that _additionally_ has the capability of aiding with parts placement than the other way around. As for right now the selected part isn't linked to that list, but maybe that makes more sense than highlighting parts?

## Pie-in-the-sky proposal

This is just me dreaming things up, feel free to discuss and intervene.

If I had a magic wand that would make things happen the whole thing might look like this:
- Parts need to be selectable in single or in bulk: maybe it makes sense to group them by value on default but allow to "unfold" the list and change the thing by which they are grouped?
- More advanced search/filter capabilites (as also mentioned in IRC)
- Additional columns for parameters like shadow/fix/do-not-place/omit-silkscreen/etc
- A button called "Placement assistant" and a way to quickly reset placed parts

I imagine the Placement Assistant would give you a fixed order in which things will be placed and you just need to step though the list with buttons called 'next part', 'skip' and 'undo' button (or arrow keys or whatever). The parts will be automatically highlighted on the board with a nice arrow like when searching with <kbd>Ctrl</kbd> + <kbd>F</kbd>. Once you click 'next part' the previous part will be ticked off as placed. When you click another part than the active one, it will use that one as a starting point for the movement. This is done till the list is exhausted. If the list is grouped by value, the Placement Assistant will move by value first and then by Refdes, if it is grouped by Refdes (which is no grouping at all) it moves by refdes number, if it is grouped by package you will place 0805 first, in groups of the same value etc.

## Additional ideas

This might be already to much, but here is more:
- Comments per part? (especially in prototypes one might change some values try some things out or need to change a value)
- Exporting of the placement list with these comments

bob-u, 2020-07-27:
This is useful.  I would like for the list selection to also highlight schematic parts, one could, for example, optimize the use of same brand/value decoupling caps for example (to find them easier within schematic).  Comment field is a great idea.  

As you say, this window addresses a mix of needs and , IMO, it is a step in the right direction.  Overall, I think that Horizon should have a selection window, where one can select objects (parts, junctions, vias, traces, nets, etc), filter them additionally by their parameters (value, manufacturer, name, etc)  to located them on the board or to make a selection without using a mouse in order to move, delete or change their parameters in bulk. 

Btw, I was wondering for a second where this new window is until I pressed the space bar and found it easily 👍

carrotIndustries, 2020-10-20:
Thanks for the input, that all more or less lines up with what I had in mind. There's one important design decision that needs to be made before we can start implementing this: Should this window be part of the board or schematic editor? The window can only do edits to the document it belongs to.

From my point of view, it'd be better off in the schematic editor as it then could be used to to things like bulk edits to the netlist such as replacing parts or setting do not place.

atoav, 2020-10-21:
I think it makes more sense if it belongs to the schematic editor, however one of the mentioned scenarios was e.g. finding all 100k resistors on a board. For this it doesn't have to have "write access" to the board, but it has to be able to highlight or update the selection not sure if this is possible tho.

Ultimately it would be desirable if it would work from both sides as this would be the most intuitive implementation from the perspective of a user who isn't deep into programming themselves. But for us that would probably mean finding a new way of dealing with two potentially out of step netlists from one place. If you are motivated enough this could be used to make horizon even greater e.g. by having a way of highlighting the differences between the netlists etc. But if you go for a more straightforward implementation I'd probably do it on the schematic side of things as well.

carrotIndustries, 2020-10-21:
> but it has to be able to highlight or update the selection not sure if this is possible tho.

Yes, that's possible using the existing IPC facilities.

atoav, 2020-10-21:
> Yes, that's possible using the existing IPC facilities.

Cool, so I think this is a pretty straightforward choice right now