horizon-eda/horizon #482: What are the rules on assigning key sequences to tools?

bob-u, 2020-07-30:
What are the rules on assigning key sequences to tools/features?  I tried to assign 'vs' to turn off snapping in schematic and feature is triggered by just pressing v alone.

carrotIndustries, 2020-07-31:
Key sequences are recognized as soon as they aren't ambiguous. As you probably don't have any other key sequence configured that starts with <kbd>v</kbd>, the first key is sufficient.

bob-u, 2020-08-01:
It appears that key assignment needs sanitization.  Here's an example where I assign ppp to draw polygon circle.  Unfortunately pp is already taken to place part.  Once ppp is entered - neither function works:

[image]

carrotIndustries, 2020-08-01:
Unfortunately, it's not really feasible to detect this when defining the key sequence as we can't tell a this point if key seqeuences could conflict.

The best we could come up with is informing the user about conflicting key sequences as they're being entered in the editor.

atoav, 2020-08-11:
> The best we could come up with is informing the user about conflicting key sequences as they're being entered in the editor.

This should give you two options:
- _Abort_ the operation
- _Unset_ the (other) conflicting key/value pair and _set_ this one

Ideally the latter option would also give you the ability to quickly change the other/conflicting shortcut afterwards, but this is a comfort nearly no software I ever used offers.

commit c4e2f8f9 (2020-08-12): tell user about conflicting key sequences

fixes #482
closed 2020-08-12 by carrotIndustries