horizon-eda/horizon #483: Feature request: object highlighting

bob-u, 2020-07-31:
I wonder if there is a technical limitation preventing Horizon EDA from highlighting objects by changing their brightness/drawing outline/etc instead of a bounding box?

example:
https://youtu.be/VNrMMW4TvuU?t=30

Btw: congratulations on the release of v1.2.0  Amazing amount of new features.

rroohhh, 2020-07-31:
If you are talking about cross-probing,  Horizon already does that, you might need to change your highlight mode or the highlight dim / lighten settings if you cannot see highlights.

bob-u, 2020-07-31:
It was probably bad example.  I am talking about highlighting part, trace, via, etc.  which is currently with a bounding box.  Here's are better examples:

Note traces change color when selected:
https://youtu.be/9hnSz1_31Wc?t=123

traces highlighted with an outline:
https://youtu.be/9hnSz1_31Wc?t=35

traces highlighted in hover mode:
https://youtu.be/TWcL0OF_eQI?t=1872

carrotIndustries, 2020-07-31:
You seem to mix up selection and highlight which are two different things in Horizon EDA:

 - The bounding boxes are used for selecting objects in preparation to do something with them.
 - Highlighting is used to for showing objects more prominently in cross probing and other contexts.

Using the bounding box for indicating selections is visually clearer than changing the selected object's appearance. Apart from that, it was easier to implement since it doesn't require modifying the rendered objects.

bob-u, 2020-07-31:
Lol, you're right.  How about we compromise and describe it as visual cue for object focus ? :)

I understand, bounding box selection being easier to implement is an important factor.  Personally I prefer object being modified to indicate selection as it feels clearer and more modern.  Also, it appears that all other EDA programs have gone that direction.  However, if it's your preference to keep the bounding box, can it be made to be less in a way (as in blocking what's around the object, for example, part values/wires in schematic or other tracks in board?  Perhaps with semi-transparent walls?

carrotIndustries, 2020-07-31:
> can it be made to be less in a way (as in blocking what's around the object, for example, part values/wires in schematic or other tracks in board? Perhaps with semi-transparent walls?

Dunno, I don't see much wrong with the bounding boxes as-is, so don't expect any significant changes in the near future.

bob-u, 2020-08-01:
Here's an example: What's the purpose of the diamond shape in the box that is covering part schema:

[image]

bob-u, 2020-08-02:
Moved to discourse

closed 2020-08-02 by bob-u