horizon-eda/horizon #488: How to package the python module

atoav, 2020-08-04:
As discussed [here](https://horizon-eda.discourse.group/t/continous-integration-generating-images-from-gerbers/15/16) it might be a good idea to bundle the [python module](https://horizon-eda.readthedocs.io/en/latest/python.html) with releases and have them installed as well so people can use them to automate things as currently building the python module requires quite a bit of knowledge about compilation etc.

The first decision might be whether the python module should be automatically installed with releases, or whether it might be better to put that in it's own installer/package..?

carrotIndustries, 2020-08-04:
> The first decision might be whether the python module should be automatically installed with releases, or whether it might be better to put that in it's own installer/package..?

The python module should be packaged separately from the main app, mostly to keep dependencies to a minimum:
 - The app doesn't depend on python and osmesa
 - The python module shouldn't depend on gtk and boost (needed for the router)

As written in the discussion, the main roadblock for properly packaing the python module is that the build process doesn't use the python-provided build utilities. I couldn't find any details on how to properly install binary python modules that were built without setuptools. @wookey Any ideas, especially related to packaging the python module as a debian package?

wookey, 2020-08-06:
On 2020-08-04 11:55 -0700, Lukas K. wrote:
> The first decision might be whether the python module should be
>      automatically installed with releases, or whether it might be better to
>      put that in it's own installer/package..?
>
>    The [...]

Debian has a fairly thorough python packaging policy:
https://www.debian.org/doc/packaging-manuals/python-policy/index.html

I don't actually know what 'the python module' is/does here, but I
agree that it probably makes sense to have it in a separate binary
package to keep dependencies down, assuming that it makes sense for
users to not have it installed.

It is tricky to do python packages that don't use setuptools, although
it must be possible. I found the easiest way was to write a setup.py
for the debian package so that it does in fact use setuptools. That
way the debian packaging is very standard and simple. But you can just
not use the python helpers and build/install the package files using
lower-level tools.

Wookey
-- 
Principal hats:  Linaro, Debian, Wookware, ARM
http://wookware.org/

carrotIndustries, 2020-08-06:
> But you can just not use the python helpers and build/install the package files using
lower-level tools.

That was my idea as well, but couldn't find a way to figure out where to put the `horizon.so` in a way that's portable since `site.getsitepackages()` returns more than one directory:
```
>>> site.getsitepackages()
['/usr/local/lib/python3.7/dist-packages', '/usr/lib/python3/dist-packages', '/usr/lib/python3.7/dist-packages']
```