horizon-eda/horizon #489: Rule check warning on Polygon Circle in Package outline

trembel, 2020-08-05:
If I draw a Polygon Circle as the only element in the "Top Package" layer, the rule checker warns with "Use polygons to draw outlines on package layers"

carrotIndustries, 2020-08-06:
I can't reproduce this, can you share the package.json where this happens?

trembel, 2020-08-06:
This is the package:

[package.json.log](https://github.com/horizon-eda/horizon/files/5036949/package.json.log)
FIY: I changed the name to `*.log`, because it is not possible to add a `*.json` to a comment.

Maybe it is connected to another bug. I experienced, that after duplicating a part (from an included pool) I'm not able to change the courtyard. See attached GIF
[image]

trembel, 2020-08-06:
The courtyard is no error. It is a polygon program. After deleting, it is possible to create a new one

carrotIndustries, 2020-08-06:
> This is the package:

Well, the thing sticking out on the top left is a line.

Since none of this was really a bug, I'm closing this issue. If there's any more need for discussion, start a topic on discourse: https://horizon-eda.discourse.group/

closed 2020-08-06 by carrotIndustries