horizon-eda/horizon #493: Please add FreeBSD to the list of systems where Horizon EDA runs

yurivict, 2020-08-07:
The FreeBSD port of Horizon EDA is now available: https://www.freshports.org/cad/horizon-eda/

Could you please add FreeBSD to the list of systems on your website https://horizon-eda.org/ and in the GitHub README file https://github.com/horizon-eda/horizon ?

Thank you,
Yuri

carrotIndustries, 2020-08-07:
Again, PRs are welcome.

carrotIndustries, 2021-01-31:
Is there anyone who at least occasionally uses Horizon EDA on FreeBSD? If not, I'm not really comfortable with claiming to support a platform where the only test is that it compiles.

yurivict, 2021-01-31:
> Is there anyone who at least occasionally uses Horizon EDA on FreeBSD?

There is no real way to know because there is no registry of users.

I ran Horizon IDA, created the PCB, downloaded the parts DB, added several components, connected them and printed the gerber output.

carrotIndustries, 2021-02-01:
> People only report back when they find issues. They don't report back just to say that they run software fine.

So far, most people ran into some sort of issue an either showed up in the issues, on IRC or on some forum. They were all using either Windows or Linux. These are also the only platforms I have at hand for testing.

That's why I'm not comfortable with claiming that it runs on FreeBSD on the landing page. Mentioning it in the docs is fine though. On that note, is `sudo pkg install horizon-eda` the right command to add to the docs?

yurivict, 2021-02-01:
Yes, ```sudo pkg install horizon-eda``` is the command to install it.

carrotIndustries, 2021-02-01:
The installation instructions now include more details on freebsd: https://horizon-eda.readthedocs.io/en/latest/installation.html#freebsd and https://horizon-eda.readthedocs.io/en/latest/build-freebsd.html

closed 2021-02-01 by carrotIndustries