horizon-eda/horizon #495: Alternate packages pointing to each other crashes the pool manager

pipatron, 2020-08-09:
While desperately trying to figure out why my board preview doesn't load the 3D model for an alternate package, I tried to set both alternate packages to refer to each other.

Shouldn't have done that, because now when I try to view either one of those packages, the pool manager segfaults:

`horizon-eda[5763]: segfault at 7fd291ffdfd0 ip 00007fd2fa0a3381 sp 00007fd291ffdfc0 error 6 in libpcre.so.3.13.3[7fd2fa099000+51000]`

I know I could have made a better report, but I thought I would just put it here when I found it, maybe you know about the problem already. I can provide a backtrace tomorrow if necessary.

carrotIndustries, 2020-08-09:
The exact cause is infinite recursion as this causes a circular dependency that isn't handled. After updating the pool, both packages in involved don't show up anymore.

I'll probably add some more checks to the pool update to tell the user that this isn't supported.

carrotIndustries, 2020-08-10:
Commits 6c8a1840b6cb929ff7ce8ec0aef7f53a6101db45 and 2c3efac5212af11fc89d1e22d575437179490989 should make circular dependencies fail more gracefully.

> While desperately trying to figure out why my board preview doesn't load the 3D model for an alternate package

Have a look at the [alt-model](https://github.com/horizon-eda/horizon/tree/alt-model) branch and let me know if this fixes it for you

pipatron, 2020-08-11:
I'm afraid the alt-model branch didn't help in the board 3D view, and now the package editor crashes when trying to view the 3D model for a single package - even the ones from the standard horizon-pool that I haven't modified. 

`Editor for '/home/pipe/share/horizon-pool/packages/connector/header/idc/2.54/2x10-vertical/package.json' exited with status 139`

Not sure how to debug that, I'm not a gdb wizard so I can't catch processes. Maybe this should be a dedicated ticket?

carrotIndustries, 2020-08-11:
Oh well, that crash was even on master and the 1.2 release. 557274b39f079a973b3fa56e4e7d6ace99d9d72b fixed it.

carrotIndustries, 2020-08-11:
I had a closer look and the actual problem of 3D models of alternate packages not showing up is a bit more involved than expected. 
A package can be associated with more than one 3D model, since that's useful for things that come in more than one color. The part then specifies which 3D model of the ones defined in the package should be used. The part doesn't know about alternate packages.

So when switching to an alternate package, we don't really have a way to figure out which 3D model of the alternate package to use. We could get around this by adding identifiers to the 3D models defined in packages so we could then match these. Any other ideas?

carrotIndustries, 2020-08-13:
The latest commits to the alt-model branch enable 3D models for the simple case if the alt pkg only has one model. Probably good enough for now?

pipatron, 2020-08-15:
First things first: The latest commit does indeed solve my immediate problem. The 3D model for my alternate package shows up. In fact, it is the same model as for the original non-alternate package.

I'm not sure how to solve your "root" problem. First I thought you could let alternate packages just inherit the model from the "normal" package, but I do think alternate packages need to have their own 3D models, even though it was not necessary in my case. I think that's more important than having different colors, but I can understand the use case for that too.

carrotIndustries, 2020-08-20:
Since both issues (the crash and no 3D model showing up) have been addressed, I'm closing this issue.

closed 2020-08-20 by carrotIndustries