horizon-eda/horizon #496: Visual Glitch: Grid spacing weird at certain zoom steps

atoav, 2020-08-13:
Okay I have the issue that on Linux all builds have weird grid glitches.
Pictures probably say more than a thousand words:
[image]  

Weirdly enough there is a spacing progression from top to bottom. Zoomed out a little more:  
[image]  

There are "sweet spots" where this effect is less pronounced:  
[image]  

Without knowing the code in question: maybe this is some sort of rounding/float precision error?

bob-u, 2020-08-13:
Does green cursor snap to the grid points as drawn?  What happens when you adjust grid settings?

atoav, 2020-08-13:
> Does green cursor snap to the grid points as drawn? What happens when you adjust grid settings?

It seems to be related to downsampling. So it only happens if I am zoomed out enough for it to start skipping the display of (inbetween) grid points. Snapping works however (it snaps onto the original grid).

carrotIndustries, 2020-08-13:
> Okay I have the issue that on Linux all builds have weird grid glitches.

Does this also happen at "reasonable" zoom levels or only when zooming out particularly far? Has it always been this way or did this happen since a particular commit or system upgrade?

The actual grid mark positioning is done [in the vertex shader](https://github.com/horizon-eda/horizon/blob/master/src/canvas/shaders/grid-vertex.glsl).

bob-u, 2020-08-15:
@atoav what are your screen resolution and grid settings?

atoav, 2020-08-16:
Okay I just tested this and it happens on Windows and Linux alike.
Grid settings are:
- `1,270 mm`
- `Square`
- zero offsets
- Fine grid modifier: `Alt`
- Grid style: `Cross`
- Grid opacity: `1.0`
- Antialiasing: `4x MSAA`

GPU is a Nvidia RTX2080M

carrotIndustries, 2020-08-16:
Try the `grid-dbg` branch and post the lines printed on stdout that look like this: `grid 3126 mod 83`

atoav, 2020-08-16:
For reference, that panel in the editor is 3U high (so ~129 mm):

## grid 8778 mod 100
[image]

## grid 4063 mod 68
[image]

## grid 7051 mod 90
[image]

carrotIndustries, 2020-08-16:
does the latest commit make a difference?

atoav, 2020-08-17:
> does the latest commit on the grid-dbg branch make a difference?

It doesn't seem like it, still very much the same. From which direction is the grid drawn? Bottom to top?

carrotIndustries, 2020-08-17:
> From which direction is the grid drawn? Bottom to top?

Yes, this code caclulates how many grid marks are there per line (`mod`) and how many there are in total (`n`). The grid mark is then rendered `n` times and the vertex shader https://github.com/horizon-eda/horizon/blob/master/src/canvas/shaders/grid-vertex.glsl#L11 then uses modulus and division to get the X and Y coordinates based on the instance id. My guess is that there's something going wrong with the modulus calculation since the X coordinates end up incorrectly.  https://github.com/horizon-eda/horizon/blob/6fbfb786172c339867845a6bf199ec404218bd0f/src/canvas/grid.cpp#L103-L106

atoav, 2020-08-17:
> EDIT: the latest commit replaces the builtin mod calculation with a custom one. does this make a difference?

This does the trick yes! No more weird grid offsets at any zoom level. Not sure what is going on tho.. some type tomfoolery maybe?

commit 8e457c1d (2020-08-17): CanvasGL: use custom modulo implementation in grid shader

the glsl mod function might be subject to floating point (?) issues on
some platforms

fixes #496
closed 2020-08-17 by carrotIndustries