horizon-eda/horizon #497: Wishlist: Save and Load design rules

pipatron, 2020-08-16:
Save/Load of design rules would be useful for two (related) reasons:

- A project with two boards always manufactured at the same fab. After setting up the rules for the first board, I want to share it with the second board.

- Using one or more favourite fab, you don't want to set up their rules all the time but keep them in a file for reuse. Reduces the work for new boards, and minimizes the risk of getting something wrong.

carrotIndustries, 2020-08-16:
I've thought about this as well, but the issue is that design rules that reference nets/net classes aren't portable between projects as-is due to UUIDs. Would it be okay to only import/export the catch-all rules?

pipatron, 2020-08-17:
Yeah, I think that's a reasonable option.

A more complicated version could be to export everything by name, if possible. On import, either just load the rules that has a matching net/class, or ask the user how to map them.

I think a catch-all load/save would give the most "bang for the buck".

commit 5f33144c (2020-12-05): add rules export/import

closes #497
closed 2020-12-05 by carrotIndustries