horizon-eda/horizon #501: Loosing odd chunks of edits mid way through a session

cyclic99, 2020-08-20:
I have had a few instances lately where I have lost edits to a schematic though subsequent edits are all there.
I am not certain, but I think that it happens when going into a sub command e.g. such as routing tracks.
If the routing goes bad one wants to go back to as before by clicking back arrow button.
However I think that if you click back arrow without fully exiting the sub command first the back arrow appears not to work, so one clicks it some more (and more and more ...).
What actually is happening is that the screen remains in the sub command but unknown to you previous edits are being deleted each click.  Ouch.
When you finally figure out how to get out of the sub command everything starts working again but a chunk of previous edits has been deleted but you restart editing unaware.
(Not much proving done as doing things like this on alive schematic can become embarrassing if ...)
I believe that back arrow aught to exit the sub command on the first back arrow click.
Thanks for all your work.

carrotIndustries, 2020-08-20:
Please provide a clearer explanation with screenshots illustrating what you're referring to.

I suppose you' referring to the undo/redo buttons in the top left with "back arrow button"? This one isn't clickable if w tool is active.

pipatron, 2020-08-27:
I don't really understand this report either, but maybe it has some relation to issue #505 I just recently reported.

cyclic99, 2020-08-28:
This #501 could be related to #505.
For the last week I have been very "particular" about trying to exit from any sub-menu-function once I have finished its use.   
(If only to avoid odd bits of track accidentality being deposited)
Since then the issue (in #501) has not re-occurred and it seems possible that the issue is related to not exiting sub menus in a timely manner.

Suggestions on how to prevent users from inadvertently have these sort of problems are :
Is there a way to make it more obvious that one is (still) in a sub-menu 
e.g. brighter and colour on its menu bar at the bottom?
Is there a way of making a visual "tell" when continuing after a delay period left in a sub menu? 
e.g. blink the (coloured) sub menu bottom bar.
Or should horizon-eda just drop out of the sub menu after a time out anyway?
I believe the Top-left back arrow and Save buttons are greyed and not highlighted when in sub-menus but should this be made more visually obvious?
Or would it be safer just to make  "back-out" and "save" available in sub menus. This could first cause the exit of the sub menu prior to carrying out the "back-out" or "save" actions.  It would be fail safe. 
(Perhaps people imagine that is what would happen anyway?  I.e. more intuitive?)
In AutoCad the screen updates during "undoing" to show very visually, so one does not undo without being aware of the undo.
I am sure that longer term users are quite happy with the way it is but for new users these ideas might be helpful to avoid problems in this area.

v1.2.1 working very well ~~ 2k tracks, 300 packages, 1.5K lines & 3k junctions - Many thanks

carrotIndustries, 2020-08-28:
>  For the last week I have been very "particular" about trying to exit from any sub-menu-function once I have finished its use.

Please explain in detail to what you're referring by this.

cyclic99, 2020-09-05:
I am unable to repeat this - It has not happened again.
It may have been fixed by 1.2.1 ?

cyclic99, 2020-09-05:
I suggested the issue is closed.

closed 2020-09-05 by cyclic99