horizon-eda/horizon #507: "Miter" and "Square" types for planes are swapped

pipatron, 2020-08-30:
World's smallest issue but still:

Planes can have a style: _Round_, _Square_, and _Miter_. For some reason, _Square_ and _Miter_ does the opposite of each other. It doesn't matter if I make the plane use the global rules or its own unique rule — they still do the opposite of what it says.

I thought it would be simple enough that even I could find the problem in the source code, but everything _seemed_ to be in the right order and connected to the right buttons.

carrotIndustries, 2020-08-30:
IIRC someone else also noted this a while back. The Plane style maps directly to clipper's join type: http://www.angusj.com/delphi/clipper/documentation/Docs/Units/ClipperLib/Types/JoinType.htm Does this match what you're seeing?

pipatron, 2020-09-01:
Indeed, that's the same behavior. By the author's use of quotes around the word _square_ I think they also didn't think it made much sense. I don't think you could find anyone who thinks this trivial example is intuitive:

[image]

Maybe the solution is to rename _Square_🡒_Bevel_ and _Miter_🡒_Sharp_ or something similar.

fruchti, 2020-10-08:
> IIRC someone else also noted this a while back.

That might've been me on IRC. I'm still for changing the naming here.

> Maybe the solution is to rename _Square_ 🡒 _Bevel_ and _Miter_ 🡒 _Sharp_ or something similar.

Sounds way better that way to me.

commit 7ec340e3 (2020-10-08): plane editor: make plane style names more clear

closes #507
closed 2020-10-08 by carrotIndustries