horizon-eda/horizon #511: GLSL downto version 1.20

Franck78, 2020-09-20:
Hello,

My laptop from 2010 cannot run Horizon. It is old but more than enough to run Gimp, Inkscape, KiCad, etc etc etc ;)
It have 
```
OpenGL vendor string: Intel Open Source Technology Center
OpenGL renderer string: Mesa DRI Mobile Intel® GM45 Express Chipset 
OpenGL version string: 2.1 Mesa 19.3.4
OpenGL shading language version string: 1.20
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 2.0 Mesa 19.3.4
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 1.0.16
OpenGL ES profile extensions:

```

It is a common issue not easily fixed. All the vertex/*.glsl need rework and probably also things like grid.cpp

Here is a project who solved the same problem. Might help.

https://github.com/projectM-visualizer/projectm/issues/168
https://github.com/projectM-visualizer/projectm/pull/399

Specs for GLSL120
https://www.khronos.org/registry/OpenGL/specs/gl/GLSLangSpec.1.20.pdf

carrotIndustries, 2020-09-20:
> OpenGL version string: 2.1 Mesa 19.3.4

That's the real reason. Horizon EDA requires modern OpenGL as in something more recent than OpenGL 3.0 or so.

>  run Gimp, Inkscape, KiCad, etc etc etc

None of these require modern OpenGL.

I'm closing this since I have no plans for supporting legacy OpenGL implementations.

closed 2020-09-20 by carrotIndustries

Franck78, 2020-09-20:
"None of these require modern OpenGL."
Does this really requires videogames perf ? Not sure of that. Trashing a perfectly capable machine ? No. Hope this choice saved you something worth it ;)

bye

carrotIndustries, 2020-09-20:
> Does this really requires videogames perf ? Not sure of that. Trashing a perfectly capable machine ? No. Hope this choice saved you something worth it ;)

When starting to implement the rendering back in late 2016 I picked what was current at that time. Using modern OpenGL made the renderer significantly easier to implement and more efficient.

I also have a laptop from around 2010 with a discrete nvidia GPU that runs Horizon EDA just fine.

Franck78, 2020-09-20:
You are lucky, because no such OpenGL for the Intel card (linux).
If I can fix some " in vec2 => attribute vec2 ", I cannot invent missing variables like gl_InstanceId or gl_VertexId.

You should  test the version and issue an explicit message. I probably won't be the only one using oldies. It will save lot of time saying no hope on this matter.