horizon-eda/horizon #512: Local environment information leaks in board.json

ellenhp, 2020-09-23:
Hey! I'm trying to check in a Horizon project and I'm having some issues with local environment information being leaked into the version controlled files. In board.json I notice this:

`        "output_directory": "/home/ellen/horizon/project",`

I'm wondering how hard it would be to move that to something I could gitignore. I know you did this with impl_meta files not too long ago so I figure you may have just missed a field? Anyway, thanks for this software it's been really nice to use :)

carrotIndustries, 2020-09-23:
If the output directory is a subdirectory of the project (though not the project directory itself), it should automatically get translated into a project-relative path.

ellenhp, 2020-09-23:
That makes sense and works perfectly. Thank you!

closed 2020-09-23 by ellenhp