horizon-eda/horizon #513: Delayed frame update on Intel integrated GPU

jbayless, 2020-09-24:
Hello,

I mentioned this on the eevblog forum but I have an update, and although there are related issues this one seems to be somewhat different, so I thought I should create a proper issue for it rather than leave discussion there.

**Related issues:**

These seem related to the Intel GPU behaviour, but they aren't reporting quite the same issue that I am.
- #9 
- #406 
- #429 

**My issue:**

It's a bit hard to describe but basically frame updates would be delayed by an action. For example, if I am placing a part, the part's location being displayed on the monitor, as I move my cursor around, would be the _previous_ location, rather than the _current_ location. The text in a text box would be whatever the previous text was, rather than the current text.

Eg. If a text box contains the value `1.250 mm`, and I want to change it to `1.350 mm`, I position the cursor behind the `.` character and hit the `del` key to delete the `2`. After doing this, the text box still displays as though it contains the value `1.250`. After taking any other action (moving the mouse, moving the cursor, etc) the box updates to `1.50`. Then I hit the `3` key, but the box still displays as though it contains the value `1.50`. Again, once I move the mouse or cursor, it updates to properly display `1.350`.

Basically, the display state always lags a bit, not by any fixed amount of time but rather until some new action takes place. Obviously this makes it very hard to get work done, although I gradually learned to cope.

**My system:**

5th gen Thinkpad X1 Carbon, Intel HD Graphics 620, driver 26.20.100.7812 (2020-01-21), 1920x1080 monitor resolution. Windows 10.

**My workaround:**

I bought an NVIDIA 2060 GPU and connected it to my laptop via external Thunderbolt graphics card adaptor. When the NVIDIA card is connected, Horizon EDA works correctly and frame updates occur as expected.

carrotIndustries, 2020-09-24:
>  but rather until some new action takes place.

That's my take as well. Horizon EDA draws like this:

 - `Canvas::render` transforms the board/schematic/etc into vertices that'll be drawn by the GPU
 - `CanvasGL::request_push` calls `queue_draw()` that'll make gtk redraw the Canvas in the next frame
 - `CanvasGL::on_render` pushes the computed vertices to the GPU and draws them

For some reason the draw operations in the last step seem to use the data from the last frame or so.

In one place, I added a workaround for this: https://github.com/horizon-eda/horizon/blob/1b931afd9460933e7cb553c46ef61453a565fb79/src/canvas/drag_selection.cpp#L328-L330

Another take is that it's related to the way Gtk renders the window if there's OpenGL, so it's probably a good idea to come up with a sample program that exhibits this issue as well, so we can open an issue with Gtk.

carrotIndustries, 2020-09-29:
Try the branch no-buf: https://github.com/horizon-eda/horizon/tree/no-buf and see if makes a difference. I don't have a windows machine with an intel gpu at hand to test right now.

jbayless, 2020-09-30:
Thanks for the update! Based on my testing today, I don't notice any difference with that branch, unfortunately.

The silver lining is that thanks to Horizon EDA, my new GPU is a company purchase. =)

carrotIndustries, 2020-12-17:
I manged to reproduce this in the gtk demo, let's see what happens: https://gitlab.gnome.org/GNOME/gtk/-/issues/3487

carrotIndustries, 2020-12-22:
I came up with a workaround and the latest windows builds now include a patched copy of Gtk containg that workaround.

Please check if this also solves the issue on your system. Not quite sure if we should close this issue since shipping a patched copy of Gtk isn't really feasible on the long run.

jbayless, 2020-12-26:
Wow, it works!! I'm amazed, Lukas... You're a wizard! Thank you!

Perhaps the thing to do is close this issue and open a new one ("TODO: Remove dependency on patched Gtk") and once Gtk integrates your patch into their mainstream release, that ticket could be closed too. What do you think?

carrotIndustries, 2020-12-26:
This issue has been resolved by shipping a patched build of gtk, further progress related to going back to upstream gtk is tracked in #539

closed 2020-12-26 by carrotIndustries