horizon-eda/horizon #515: Moving ancillary data out of design files

carrotIndustries, 2020-09-24:
#512 got me thinking if it might be worth moving ancillary data such as rules and export setting out of the design files (`board.json` and friends)

In the case of a board, that'd be:
 - Rules
 - Fabrication output settings
 - Colors (solder mask + substrate)
 - PDF export settings
 - STEP export settings
 - PnP export setting

I see several advantages:
 - Less coupling in the code
 - Rules, etc are extracted from the board while opening it and only written back for saving, so they don't really need to be part of the board
 - Prettier diffs?
 - Fab. output setting can be copied from one project to another if stored in their own file

This would be a somewhat breaking change as older version wouldn't be able to read the rules etc, so we need something similar to #280 to be in place before implementing this.

Any other thoughts?

bob-u, 2020-09-25:
Slightly off-topic: Speaking of changes, have you considered using HJSON in place of JSON?

https://hjson.github.io/

Pitrsek, 2020-09-25:
Having a fab settings and rules as separate files(and ability to save/load) would be pretty neat.

unrelentingtech, 2022-01-15:
Would be nice to be able to not include export settings (especially last exported file path) in a public project. These can be kind of a surprising privacy leak, tbh.

RX14, 2022-01-15:
On the other hand, I usually have the export directory specified relative to the project directory, so in these cases it makes sense to include.

unrelentingtech, 2022-01-15:
Sure, it just should be immediately obvious to the user that they are saved in the project.