horizon-eda/horizon #517: top-right menu doesn't always shows

rhaamo, 2020-09-29:
That's a weird one, the top-right menu doesn't always shows.
Sometimes it half shows, half transparently, or not, and hovering it makes the items appears.
[image]
[image]

I'm on ubuntu 20.04 LTS, amdgpu 20.20-1089974 for drivers, dunno what else can cause that, I've built horizon from git tag `v1.2.1`.
Latest master also have that issue.

carrotIndustries, 2020-09-29:
Probably a gtk / gpu driver / theme bug. You might try changing to the default (Adwaita) theme.

Milek7, 2020-10-17:
I filed https://gitlab.freedesktop.org/mesa/mesa/-/issues/3657