horizon-eda/horizon #520: Delaunay triangulation is a significant performance bottleneck on large board editing

eqvinox, 2020-10-11:
Hiya, this is a "tracking" / "notify" type of bug to note down awareness and investigation of an issue.

I've recently created a larger board (320x150mm, BGA parts) in Horizon and noticed the board editor is slowing down significantly, especially when drawing wires.  My board is still workable-ish, but delays are bordering on the intolerable (1-2s lockups.)

After some investigation with `perf`, the major culprit seems to be the recreation of the Delaunay triangulation to calculate airwires with >80% of CPU time being spent there.  The worst offender is this loop: https://github.com/horizon-eda/horizon/blob/master/3rd_party/delaunay-triangulation/delaunay.h#L61-L77 

I've tried/hacked two things:

* I tried creating and "index" for triangles to find relevant triangles fast instead of walking all of them.  This kinda-sorta helped, but I broke things (I couldn't find a good data structure for this kind of 2D lookup and cobbled together some trees but didn't do it right...)  Tried fixing it, and when it was correct again it was just as slow as before.

* I split off the triangulation into separate threads for each net.  Theoretically, the triangulation is independent for each net, and this does indeed help performance a little.  Again, though, I didn't do it completely right and it works 99% of the time ... until it deadlocks somewhere.  My understanding of C++ STL datastructures is not enough to know where there might be races in there (there certainly is one /somewhere/...)

In the long run, I think the only good fix is to not recalculate the triangulation all the time and instead remove and re-add points incrementally.  This is unfortunately not very well covered by existing Delaunay libraries (point removal is a bit fringe...), and it also needs deeper-going changes in Horizon to keep track of which points to remove and re-add while doing other operations like moving things around.

Anyway I just wanted to dump this information here so it's not a waste of my time investigating the issue, and also note down what I tried.  Cheers!

carrotIndustries, 2020-10-11:
Thanks for pushing the boundaries of Horizon EDA! Algorithms is one thing where we don't have to reinvent the wheel and can take inspiration from KiCad, for example https://ohwr.org/projects/cern-kicad/wiki/ratsnest-gal

IIRC KiCad uses a simpler ratsnest algorithm during interactive move that just finds the nearest point for the nets contained in the selection. KiCad also uses another triangulation library that could be faster than the one we're using.

>  but delays are bordering on the intolerable (1-2s lockups.)

Are these occurring after ending a tool?

In general, there's substantial scope for improvement in terms of speed that mostly boils down to the editor not knowing what a tool did so it has to recompute more than strictly required. If more people start Horizon EDA for more complex boards it's probably worth looking into improving the performance.

Is rendering itself (zooming&panning) getting laggy?

eqvinox, 2020-10-13:
> > but delays are bordering on the intolerable (1-2s lockups.)
> 
> Are these occurring after ending a tool?

Yes, it's most apparent after moving a package, but also after each click to draw a wire (and for some reason also when *starting* the route track tool, not sure what that's about...)

> Is rendering itself (zooming&panning) getting laggy?

No, that's working well comparatively.  (It's always a bit sluggish on my system but that's a problem I have with my desktop overall - an Intel UHD 620 / i7-8550U is quite underpowered for a 4K screen.)  Either way I'm not noticing a difference in zoom/pan between an empty board and my large design.

carrotIndustries, 2020-10-21:
> Yes, it's most apparent after moving a package, but also after each click to draw a wire

Could you share the project that causes the sluggishness? That'd really help me to get a better idea for possible avenues for optimisation.

carrotIndustries, 2020-11-11:
Due to other issues with the existing delauay triangulation library, I'm in the process of moving to delaunator (claims to be fast, also used by kicad). You can try it out on the `airwires-delaunator` branch.

phkahler, 2021-01-10:
> * I tried creating and "index" for triangles to find relevant triangles fast instead of walking all of them.  This kinda-sorta helped, but I broke things (I couldn't find a good data structure for [...]

We have similar issues in SovleSpace where some algorithms search an entire list of items to find those nearby, for every item in the list. This is O(n^2).  I have a fairly efficient spatial index that I wrote long ago. I can dig that up if it might help. I needs cleanup, but can find nearby items in O(n * log(n)) time. The index can also be built with that time complexity too so it scales well all around.