horizon-eda/horizon #521: Idea: Pin alternation

katyo, 2020-10-12:
If I understanding things right, alternate pin names used only as hints for schemata designers.

In practice to get features like DRC working with alternative pins assignments currently I spend time to create symbol with project-specific pin names and directions.

I assume that is may (and possibly should) be automated.
Say, we can introduce so called "configurations" (or shortly configs) to units (and/or entities maybe).
Each config defines which pins should be redefined and how (by overriding pin name and direction).
Schemata designer can view available configs for unit and select needed. As a result unit and symbol will be tuned accordingly.

Of course, in some projects several configurations may be used. As an option this may be solved allowing select multiple configurations at once (i.e. using checkboxes instead of radio buttons). In that case DRC rules checker must checks all possible combinations.

Configuratons may have conflicts, say, you cannot use some port as output when you already enabled analog input or alternative function. The conflicts can be checked by intersecting altered pins. In similar situations unavailable configurations should be diabled for choicing by designer.

Just for example:

* some MCU unit:
  - configs:
    - Port A1:
       - [ ] Hi-Z
       - [ ] Input floating
       - [ ] Input pull-up
       - [ ] Input pull-down
       - [ ] Output
       - [ ] Output opendrain
       ...
    - Analog input 1:
       - [ ] Disabled
       - [ ] Enabled
       ...
    - Serial interface 1:
      - [ ] Disabled
      - [ ] ALT FN1
      - [ ] ALT FN2
     ...
    - Debug interface:
      - [ ] SWD
      - [ ] JTAG
      - [ ] Disabled

carrotIndustries, 2020-10-12:
> In practice to get features like DRC working with alternative pins assignments currently I spend time to create symbol with poject-specific pin names and directions.

I noticed that deficiency as well a while back. From my point of view, the best moving forward is this:

 - Enhance the alt. names in units to include direction for each one
 - Use UUIDs for alt names rather than index-based referencing. To provide a transition mechanism for index-based alt. names, these will get assigned version 5 UUIDs based on index
 - Since we have user-definable pin names in the schematic, add user-definable pin direction as well.

The configuration thing you described seems like it could get arbitrarily complex. Figuring out pin mapping is best done using the vendor-provided tools such as STM32CubeMX.

katyo, 2020-10-12:
Sounds reasonable. Usually simpler solutions works better.

But it not excludes ability add configs over this in future. And of course we can import it from vendor's db.

carrotIndustries, 2020-10-18:
With #280 solved, we can implement this after the next release

commit 496129da (2022-01-18): use UUIDs for alternate pin names

also add per-name direction

closes #521
closed 2022-01-18 by carrotIndustries