horizon-eda/horizon #523: Switch to Meson build system

pwnorbitals, 2020-10-18:
Follow-up to https://github.com/horizon-eda/horizon/pull/111
This is the tracking issue for the switch to the Meson build system (https://mesonbuild.com/)

pipatron, 2020-10-20:
It was extremely refreshing to find a project where the developer knew how to code a makefile instead of inventing a whole lot of extra complexity, and Horizon is the easiest tool I've built in years. Could you enumerate the actual problems to solve here?

Sean-McVeigh, 2021-01-05:
@pipatronSimplifying cross-platform development and making it easier for the community to contribute to is an absolute necessity for a project like this. The only other viable FOSS alternative being KiCad, and frankly KiCad's library management is so primitive given current times. I'd really like to see more adoption of the Horizon's pool model.

carrotIndustries, 2021-01-06:
> but its way more complicated than it needs to be 

The last time someone had a go at cmake (#111), it didn't end well and using CMake caused more complexity than it did good. Apart from that, Meson seems to be more popular among gtk-based projects.

> These should at least be submodules

These don't play well with `git archive`/ github's automatic release tarballs and there doesn't seem to be a canonically accepted first-party solution for creating tarballs from repos with submodules.

> Simplifying cross-platform development

The current Makefile works on linux, msys2 and freebsd

> and making it easier for the community to contribute

Most changes should only require adding/removing source files. I don't see how CMake/meson will make this easier.

> to push adoption

Most users probably don't really care as long as there's an easy way to install it. No one ever told me that this project was particularly difficult to build.

> I'd really like to see more adoption of the Horizon's pool model

How does this relate to the build system?

fk0815, 2021-01-15:
> > These should at least be submodules
> 
> These don't play well with git archive/ github's automatic release tarballs and there doesn't seem to be a canonically accepted first-party solution for creating tarballs from repos with submodules.

This `3rd_party` directory was a blocking point when I wanted to add horizon to opensuse repositories. It is not clear which licenses are used for those packages and if they are compatible. Also the version tracking is not given.

On the other hand I know the `git archive` limitation and I have no better idea how this can be handled.

carrotIndustries, 2021-01-15:
> This 3rd_party directory was a blocking point when I wanted to add horizon to opensuse repositories. It is not clear which licenses are used for those packages and if they are compatible. Also the version tracking is not given.

Good to know, debian people weren't bothered by this.

I've started to update the list in the README file: https://github.com/horizon-eda/horizon/blob/3rd-party-info/README.md#included-third-party-software

I'll update/figure out the versions after the 1.4.0 release (end of January).

carrotIndustries, 2021-01-30:
> I'll update/figure out the versions after the 1.4.0 release (end of January).

As promised, that list is up-to-date now: https://github.com/horizon-eda/horizon/blob/master/README.md

fk0815, 2021-01-31:
Maybe it is worth to open a new topic for that, but how is guaranteed that the `3rd_party` modules are compatible when the sources are copied into the project?

carrotIndustries, 2021-01-31:
> but how is guaranteed that the 3rd_party modules are compatible when the sources are copied into the project?

Most libraries are copied as-is, some like the KiCad router and s-expression parser required some local modifications. Debian for example applies a patch to the makefile that replaces the bundled dxflib with the one packaged by them.

In general, it's expected that the `3rd_party` directory is used by downstream packagers as-is.

frmdstryr, 2023-08-29:
Looks like this was done in https://github.com/horizon-eda/horizon/commit/f905862e52897dd0cb257bd4721e1fbf4ba713bb and this can be closed?

carrotIndustries, 2023-08-29:
> Looks like this was done in https://github.com/horizon-eda/horizon/commit/f905862e52897dd0cb257bd4721e1fbf4ba713bb and this can be closed?

Thanks for the hint.

closed 2023-08-29 by carrotIndustries