horizon-eda/horizon #524: Feature request: integration with jlc components database

bob-u, 2020-10-20:
Extend parametric search for resistors and capacitors to include jlc vendor part list, and include vendor specific features such availability, pricing, basic or extended assembly.  Similar to this website:

https://yaqwsx.github.io/jlcparts/

carrotIndustries, 2020-10-20:
The stock information column in the part browser is backed by the `StockInfoProvider` class that could also be implemented with a backend to the json files generated by that site.

This however doesn't make the parts appear out of thin air.

isometric, 2020-12-23:
this seems scrapable but that's going to be hell to review & dedup with existing
https://github.com/xtoolbox/lc_kicad_lib