horizon-eda/horizon #525: exception thrown in tool_begin of Route track

Pitrsek, 2020-10-24:
I tried to route my board today and was greeted with:
Seq | Level   | Domain                  | Message; Detail
0   | Info    | Block                   | loading block b13c5a15-6c9d-497c-ba16-e2e4ea29bd03;  (startup)
1   | Info    | Board                   | loading board 993142f7-5358-40f3-b8a4-741661e0a8f3;  (startup)
2   | Critical| Core                    | exception thrown in tool_begin of Route track; invalid pad polygons: 0

Reboot did not helped, I haven't a slightest clue as to what has changed.To my knowledge my setup remained the same. Happens when i press x to route. 

Cheers,
Pitrsek

Pitrsek, 2020-10-24:
I realized that I have updated a package footprint. When I delete it, routing works.

carrotIndustries, 2020-10-24:
> When I delete it, routing works.

Please share that particular package so I can reproduce it.

Pitrsek, 2020-10-24:
Here you go. 
[CP-40-7V2.zip](https://github.com/horizon-eda/horizon/files/5433203/CP-40-7V2.zip)
I think i know what is causing the issue - package contains pads that do not have copper layer. I used those to create the shape, instead of drawing with curves.

carrotIndustries, 2020-10-24:
The zip missing the padstacks.

Pitrsek, 2020-10-24:
I've zipped whole project(@ your mail), hopefully it will work as everything should be in project pool cache.

carrotIndustries, 2020-10-24:
> I've zipped whole project(@ your mail), hopefully it will work as everything should be in project pool cache.

Haven't received anything yet.

commit e43a8309 (2020-10-24): router: don't care about padstacks with zero polygons

fixes #525
closed 2020-10-24 by carrotIndustries