horizon-eda/horizon #528: Placing via on pad centre doesn't connect to pad

RX14, 2020-11-07:
When placing a via in a pad, you have to draw a trace to the pad centre or horizon thinks there's no connection. This is fine, but when placing a via directly on the centre of a pad, the track has to be zero length. I've worked around this by drawing 0.001mm traces to my vias in pads, but this isn't ideal.

I suggest either considering any via touching a pad in the same net to be connected - or alternatively, consider vias which are directly at the pad centre to be connected.

carrotIndustries, 2020-11-11:
See if the `airwires-delaunator` branch makes a difference.

>I suggest either considering any via touching a pad in the same net to be connected

Unfortunately, we don't really know how a pad looks like before rendering it, so we'll have to live with the connection only being made by the via being exactly at the pad's origin.

RX14, 2020-11-12:
> Unfortunately, we don't really know how a pad looks like before rendering it, so we'll have to live with the connection only being made by the via being exactly at the pad's origin.

I'm okay with this!

The "false" airwires are gone when I build with the `airwires-delaunator` branch, thanks a lot!

closed 2020-11-15 by carrotIndustries

RX14, 2020-11-23:
This still seems to fail actually, on more complex scenarios...

[image]

Note airwires on GND pads on the right hand side and USB_D+

The B12 GND pad has figured itself out correctly though

carrotIndustries, 2020-11-23:
Can you share a minimum working design to reproduce this?

commit 3d5710ba (2020-11-15): use delaunator for airwire calculation

fixes #528