horizon-eda/horizon #529: Feature request: Virtual bridge to divide two nets connected to each other.

joshuawins, 2020-11-09:
I wish there is a virtual bridge to split the nets actually connected to each other.
For example, it can be used to split analog/digital section that is connected through narrow polygon path.
And another example will be 4 traces to connect the PT100 temperature sensor. Each pair of wire, one is to measure voltage and the other is to follow current.

joshuawins, 2020-12-03:
"PCB " would be a great solution. It also can be used as a PCB antenna. 
How about just make some option for some trace or polygons of a part to be left on a gerber output?

jbayless, 2021-06-18:
This would be a great feature. I actually a "short circuit" part that's just two terminals connected by a trace, specifically so I could add multiple names for a single net. I find it helpful organizationally to add extra names when a single net has distinct functions (eg. OUT1 might be the same trace as IN2 but the naming keeps the logic clear). Especially when it comes to things like LOGICGND, POWERGND, and SIGNALGND, which might all connect to the same copper plane but still need to be kept separate to avoid ground loops and such.

carrotIndustries, 2022-03-23:
implemented as net ties in 9eaad29ea0b5834c061d43ee52a2215bdead6beb