horizon-eda/horizon #530: TextBoxes Lag

viniciusmiguel, 2020-11-14:
On Windows 10 2004 Build 19041.572 the TextBoxes are presenting Lag between the keypress and visualization.

Tested with last release 1.3.0 on a clean installation.

carrotIndustries, 2020-11-14:
Which ones exactly? Are you using an intel gpu? There've been reports about things not working optimally with them: https://github.com/horizon-eda/horizon/issues/513

viniciusmiguel, 2020-11-14:
Intel Xeon E-2176M
64Gb RAM
NVIDIA Quadro P4200

All the text boxes.

viniciusmiguel, 2020-11-14:
I disabled the internal integrated graphics and the lag disappeared.
I can confirm there is an issue with intel integrated graphics.

carrotIndustries, 2020-12-17:
I manged to reproduce this in the gtk demo, let's see what happens: https://gitlab.gnome.org/GNOME/gtk/-/issues/3487

carrotIndustries, 2020-12-22:
See https://github.com/horizon-eda/horizon/issues/513#issuecomment-749833082, this should also apply to your issue. Can you check if this workaround works on your system?

carrotIndustries, 2020-12-26:
This issue has been resolved by shipping a patched build of gtk, further progress related to going back to upstream gtk is tracked in #539

closed 2020-12-26 by carrotIndustries