horizon-eda/horizon #531: Exception is thrown when I click on [Update PR]

jue89, 2020-11-14:
How to reproduce:

* I installed Horizon 1.3.0 using AUR.
* I removed existing ~/.config/horizon
* I removed existing ~/Documents/horizon-pool
* I started Horizon and downloaded the default pool
* I logged into my Github account using to Pool manager
* I took one of my PRs and clicked that "Update PR" button

That exception is shown:

[image]

jue89, 2020-11-14:
Well, after thinking about it ... it looks like, I need to add my fork to the remotes of the pool's git checkout?!

carrotIndustries, 2020-11-15:
> Well, after thinking about it ... it looks like, I need to add my fork to the remotes of the pool's git checkout?!

Not you need to do it, but the pool manager needs to (which it does now).

When implementing the PR update thing, it must have slipped my mind that the user might have created the PR with another copy of the pool, so I assumed the branch and remote to be there.

commit 88ca9811 (2020-11-15): pool remote box pr update: don't crash if there's no local branch

fetch remote branch if it's not existing locally

fixes #531
closed 2020-11-15 by carrotIndustries