horizon-eda/horizon #532: Gl Error 1285 when trying to add 3D model

allderdicerobotics, 2020-11-19:
[image]

This happens whenever I try to add a 3D model.

This is the file I am using:
[User Library-VQFN-20 3x3 0_45mm v2.zip](https://github.com/horizon-eda/horizon/files/5569399/User.Library-VQFN-20.3x3.0_45mm.v2.zip)

carrotIndustries, 2020-11-20:
I've had other people report the same error also on nvidia gpus on windows. Unfortunately, I haven't been able to reproduce this so far. Try https://ci.appveyor.com/project/carrotIndustries/horizon/builds/36411382 and post the error message. That should help pinpointing the bug.

allderdicerobotics, 2020-11-27:
Well... the bug no longer occurs in that build. As far as I can tell step files import fine now.

carrotIndustries, 2020-11-30:
> Well... the bug no longer occurs in that build. As far as I can tell step files import fine now.

Does that also apply to the latest builds from master?

allderdicerobotics, 2020-12-02:
> 
> 
> > Well... the bug no longer occurs in that build. As far as I can tell step files import fine now.
> 
> Does that also apply to the latest builds from master?

Latest build from master is good too! No more error. Thank you!

carrotIndustries, 2021-01-21:
I have a feeling that we're onto something similar to #550

commit 438e65c3 (2021-01-21): Canvas3D: don't render vertices not pushed

should fix #532, #223
closed 2021-01-21 by carrotIndustries