horizon-eda/horizon #533: horizon-imp fails to start

jue89, 2020-11-20:
*Environment: Arch Linux; Horizon EDA 1.3.0 installed using AUR; Freshly initialized config and pool.*

When I try to edit a symbol, a padstack or a package, `horizon-imp` fails to start:

```
spawn /usr/bin/horizon-imp -y /home/jue/Documents/horizon-pool-foo/symbols/diodes/tvs/2ch-feedthrough-esd-protection.json 
/usr/bin/horizon-imp: symbol lookup error: /usr/bin/horizon-imp: undefined symbol: _ZN21STEPCAFControl_Writer8TransferERKN11opencascade6handleI16TDocStd_DocumentEE25STEPControl_StepModelTypePKc
end proc 99629
```

The application shows a notification that the editor has exited with exit code 32512.

jue89, 2020-11-20:
Hmm, I cleared the pacman cache and reinstalled (resp. recompiled) horizon-eda. It fixed this issue for me ...

carrotIndustries, 2020-11-20:
Seems to be related to a recent opencascade upgrade in arch. I just updated opencascade as well and got this message:
```horizon-imp: Symbol `_ZTV21STEPCAFControl_Reader' has different size in shared object, consider re-linking```

Seems like the new version of opencascade requires rebuilding Horizon EDA.

closed 2020-11-20 by carrotIndustries