horizon-eda/horizon #545: [UX] Feedback on Decals feature

atoav, 2021-01-09:
As someone who often uses the dxf-import feature to create graphics for PCBs I really like the new Decals feature. Not having to select a hundred vertecies by hand is a huge improvement. The graphics I import are often complex and hand-drawn (think comics and such); my workflow is described [in my blogpost here](https://modular.atoav.com/blog/create-a-panel-using-illustrator-and-horizon-eda).

A few minor issues I have still with decals after having used them a lot over the past days:

- When activating the `place decal` action and selecting a decal, pressing <kbd>E</kbd> doesn't flip the decal. This works for the `place part` action and should also work here.
- Even when my decal just consists of shapes on the silk screen layer I have to select it on the Top- or Bottom Copper Layer. This is very counterintuitive and should be fixed. E.g. imagine having Top Copper invisible to focus on the silk screen and not being able to select the silk screen decal. When I first encountered this I thought it was a bug. My expectation would be that I would be able to select the decal from every layer that has shapes on it, and _not_ be able to select it from layers without anything on it. This is also a problem because one doesn't intuitively know where the origin of the decal _is_. That means one has to select the silkscreen decal on top of all the existing traces etc.
- Scaling factors are always rounded to one digit (already mentioned this one in irc)

Other than that it is a great feature.

carrotIndustries, 2021-01-09:
Thanks for the feedback:

> pressing E doesn't flip the decal.

Fixed in 5f7a85d334b87867363d7e8eb17ff5db554a1ff0

> Even when my decal just consists of shapes on the silk screen layer I have to select it on the Top- or Bottom Copper Layer

Fixed in cb34b24e7b70e11df2e8f9333822e9c33bb09dbf

> Scaling factors are always rounded to one digit (already mentioned this one in irc)

Should have already been fixed in 51478cc44ee735e518e81386851cae0e39f65dc5. Since e8cc033b90476a3645896605ae40b846bd1f3c98, the scale tool also scales decals.

closed 2021-01-09 by carrotIndustries

atoav, 2021-01-09:
That was quick!