horizon-eda/horizon #546: Artwork for the project manager?

carrotIndustries, 2021-01-09:
The project manager window (as well as the start page) can look a bit bland. How about spicing it up with some artwork @atoav?

Rather than a colored image as shown in the mockup, I was thinking about something monochromatic. If we can't come up with something that works well with light and dark themes, programmatically recoloring is also a viable option.
[image]

atoav, 2021-01-10:
Sure we could do that. The hardest part will be to do the fluid layout:
- Image needs to be a fixed distance from top, unless the image height is exceeded, then it should stick to the bottom
- Image needs to fill the whole lower section of the screen in width

The kind of image to choose depends indeed a little bit on the themes used as you already mentioned. What kind of effects can you run on the images? I think Hue-Rotation might already help. Maybe I could also come up with some new thing that is a little bit more subtle and less attention-grabbing?

carrotIndustries, 2021-01-10:
> Image needs to be a fixed distance from top, unless the image height is exceeded, then it should stick to the bottom

I was thinking more of of the image anchored to the bottom.

> Image needs to fill the whole lower section of the screen in width

Perhaps we can come up with something simple that tiles.

> What kind of effects can you run on the images?

The easiest approach would be if the artwork itself was greyscale / black with alpha. The code would then replace black with the theme's foreground color.

> Maybe I could also come up with some new thing that is a little bit more subtle and less attention-grabbing?

How about only the grid as it's visible on the right side of the footer?

frmdstryr, 2021-04-21:
The picture looks great! 

What about also adding some project "stats"?  Like part count, board dimensions, etc.. Maybe notes or last updated/dates.