horizon-eda/horizon #547: Feature request : nested / hierarchic schematic sheets

pwnorbitals, 2021-01-10:
@carrotIndustries : "Not very high on my to-do list"
I think this would be very useful and this is a feature KiCad does extremely well. It helps make schematics clearer and can help dealing with separation of concerns and modularity when teams work on a project.

anuejn, 2021-03-08:
It would also be cool if we could have something like parameters for subsheets. 
So that I can have a generic DC/DC sheet that can then generate multiple voltages depending on the resistor values I set

Pitrsek, 2021-07-05:
I'll add myself to the queue for hierarchic. It makes schematics easier to comprehend, allows for nice design reuse...

carrotIndustries, 2021-07-17:
Good news, I decided to start adding support for hierarchical schematics. Might take a while for it to become presentable, but it's making good progress so far!

carrotIndustries, 2021-07-23:
If anyone wants to take a look: https://github.com/horizon-eda/horizon/tree/hierarchy-preview

Please be aware that it's still in a rather unfinished state and many things flat out don't work or just crash if looked at the wrong way. There also were some incompatible changes in the project files, so only try it on new projects.

In the schematic properties window, click plus to add a new block. The newly-created block should appear in the sheet list. To create ports from nets, use the manage ports tool. By selecting the block itself rather than a sheet of it, the schematic editor transforms into the block symbol editor. Once there's a symbol, use use the "place block" tool to instantiate it.

Annotation, PDF export (the hierarchy is reflected in the PDF TOC), BOM export and search should work to some extent.

The next big step is implementing netlist flattening to give the board editor something to work with.

I'm still not quite sure how to handle power nets and net classes. I see two options there:

1.  Power nets and net classes are global and are automatically copied into non-top blocks
2.  Power nets and net classes are managed on a per-block basis and get matched by name during netlist flattening.

To me, option 1 seems to be the right one as it frees the user from having to manage power nets and net classes in each block individually and I don't see much value in allowing block-local power nets and net classes.

Pitrsek, 2021-07-25:
Option 1 seems good. I could not come up with a design(with my workflow and schematics habits) that option 1 would not allow for. If I have a separate power nets, I want them to be named differently, so it is explicit to the person reading the schematics. If I have 3.3V net in one sheet, and 3.3V in another one, I assume that they are connected together.

carrotIndustries, 2021-08-08:
It's nearing completion, check out the linked branch.

closed 2021-09-08 by carrotIndustries

bcattle, 2021-12-25:
I apologize if this is not the correct place to ask this, feel free to redirect me. Is there a way to have per-block variables? For example, you instantiate an LDO block three times but set the values of the voltage divider resistors to different values in each block to generate different output voltages? 

A potential workaround in this particular case would be to expose those voltage divider pins as ports, and connect the resistors external to the block.

carrotIndustries, 2021-12-25:
> I apologize if this is not the correct place to ask this, feel free to redirect me.

https://horizon-eda.discourse.group/ is the place to be

commit 7d075781 (2021-08-09): add support for hierarchical schematics

closes #547
commit 25f855ae (2021-08-11): add support for hierarchical schematics

closes #547
commit 16b925a3 (2021-09-08): add support for hierarchical schematics

closes #547