horizon-eda/horizon #548: Better Symbolic Application Icon + Application Icon Ideas

frhun, 2021-01-11:
So, i played with the application icon a bit.
I like the GNOME style of icons, which try to have icons be a strong metaphor for the actual utility of the program. As Horizon fits in the GNOME ecosystem, i thought it would be nice to have horizon conform to that style.
As i already did the work, i thought it couldn't hurt to share it.

[image]

I incorporated the current logo as not to diminish the visual identity, but added elements that make the function of the program clear from the icon alone.

Even if you don't like the thought of changing the full-color icon, the current icon doesn't work well in the GNOME top bar.
GNOME uses the symbolic version of icons for the top bar, if available. So maybe at least the symbolic icon version could be of use.

This is how the current icon looks in the GNOME top bar:
[image]

bob-u, 2021-01-11:
Speaking of icons, it would be nice if Horizon windows had a icon representation that hinted their function in the tack/window bar.

frhun, 2021-01-11:
As to @bob-u s suggestion, i made a few Icons in the Windows style. While they all share the same logo (just executed differently) as a common element, it is questionable if diverging app icons on different platforms are a good idea.

[image]

carrotIndustries, 2021-01-11:
@atoav What are your thoughts on this?

atoav, 2021-01-11:
Hey, I like the ideas. I also had the idea that more icons would be needed (especially to seperate the different editors visually in the task bar). So I welcome the effort @frhun put into this!

I really like the minimalist variant of the logo in the last post and would have nothing against using it. Slight mutations in graphical style show that a project is alive and kicking : )

For the Gnome-Style Icons, I feel it needs a bit fine tuning and maybe combining it with that new logo style would look really good. I don't see any issues having different icons on different plattforms – as long as it feels native I'd even see this as a plus (because it shows that we value the plattforms).

For the Windows Icons, I like the board editor one, and I'd love the schematic editor to have a similar minimalist feel to it. I am not sure what the gray block-gradient one represents (project window?), but it looks good and recognizable. The last three icons stil feel a little bit rough to me, and are probably a bit easy to confuse with each other.

@frhun feel free to open a pull request in the [horizon-artworks repo](https://github.com/horizon-eda/horizon-artwork) once you feel satisfied, then we can continue discussion there – what do you think?

frhun, 2021-01-11:
The icon with the gradient was meant to look similar to the excel icon (as a prime example of a platform native application), maybe for the parts browser.  I agree that it doesn't really convey any real function.
I think the icons could use a bit more color, just to be more recognizable. The distinct color scheme of the schematic editor really lends itself to that, now that i think about it.
I'm not really sure how to convey the part browser though. (without too much visual clutter)
I will play around a little more, and we will see if it works.

frhun, 2021-02-02:
So.. i didn't want to open a PR jet, because i'm still pretty undecided what version i like most. So maybe let me know what tendencies you like most, or just play around with it in Inkscape yourself.

[image]
[image]
[image]
[sources.zip](https://github.com/horizon-eda/horizon/files/5914838/sources.zip)

carrotIndustries, 2021-02-05:
@atoav your thoughts? Should these icons be used for the various editor windows? In that case, we need to make sure that they still look okay and are recognizable when rendered at minuscule sizes, such as 16×16 or 32×32. Gtk recommends providing 16, 32 and 48 px versions.

How about omitting the app logo from the editor icons? That'd be in line with the editor window titles not showing the application name as well.

atoav, 2021-02-05:
My thoughts: I really like the direction this is going. I especially like the idea of using color for the window icon differentiation (e.g. yellow for parts editor and green for schematic editor). What is still a bit unclear to me is, whether these are variants of the same icon or if some of these are meant to be for the parts editor, pool editor or others. Overall however, great work!

I share @carrotIndustries concern that these icons might not work as well in a smaller scale (especially at 16×16px the small horizon logo might geht too small).

Took the freedom to give it a try myself. I mainly took inspiration from your work, and tried to change the form factor by making the horizon logo fill half the icon at all times, I additionally finetuned some of the colors and other things to my (very subjective) taste:
| What                                      | Icon(s) | Comment |
| ----------------------------------------- | ------- | ------- |
| Application Logo/Main Icon/Project Window |  [image] |  Same, but the blue is a bit "paler" |
| Window: Schematic                         |  [image]  |  Went for white, because it feels most logical  |
| Window: Board Editor                      |  [image] |   Very similar to your thing a bit more complexity in the tracks so it won't become too abstract      |
| Window: 3D View                           |  [image] | This one still needs work, but I took the default background gradient as a mnemoic anchor people already know |

They also work quite ok at 16x16 (even without the usually necessary adjustments of line weights, complexity and pixel alignment): 
[image] [image] [image] [image]

As you see, at that scale color acts as the main differentiator. Also we still should be aware that this window belongs to horizon.

frhun, 2021-02-06:
> What is still a bit unclear to me is, whether these are variants of the same icon or if some of these are meant to be for the parts editor

Both, in a way. The idea is that the main icon will represent the Project as a whole, across platforms and sub applications.
Then every platform gets it's own variation that fits in with the design language of that platform.
For example the first two variations are for the GNOME platform, which relies on physical metaphors for it's icon design.
Additionally on GNOME icons are generally not rendered smaller than 64x64, anything smaller than that uses a separate symbolic version. Also the GNOME environment has no use for different icons for different sub applications because there is no taskbar.

Windows (and many smaller Linux DEs for that matter) on the other hand rely on a taskbar for switching windows, where an easily identifiable icon for every window in the taskbar is helpful. So this is what the different designs on the bottom are for.

I agree that the idea of having a small version of the main logo with every icon doesn't really work when the icons get smaller.

I really like your icon for the 3D View.

bob-u, 2021-02-07:
My 5c: I like the symbolic icon the most because shares the simplicity of an electronic symbol.   Maybe use that and color to indicate function?

frhun, 2021-06-12:
> My 5c: I like the symbolic icon the most because shares the simplicity of an electronic symbol. Maybe use that and color to indicate function?

So this is my try at executing that idea.

The symbolic and full hicolor icon on Linux would be additional to that. The per window icons aren't really applicable to linux (GNOME) anyways.

[image]
[image]