horizon-eda/horizon #549: Tutorial for creating a part from scratch

pwnorbitals, 2021-01-15:
I think having an illustrated guide showing the full process of how to create a new part, including creating symbols/units/entities and padstacks/packages could be useful. I'm honestly a bit lost as to what the process should be, and even though I am starting to wrap my head around it by trying to understand open pull requests, I might not be the only one in this case

frmdstryr, 2021-01-22:
Have you seen the docs section on the pool?  https://docs.horizon-eda.org/en/latest/pool-why.html

It is a bit complicated/overwhelming at first but the design makes sense.

pwnorbitals, 2021-01-23:
Yeah, I have. It is completely overwhelming and is not beginner friendly, but I agree the design makes sense. That has nothing to do with a tutorial, though