horizon-eda/horizon #550: Crash on selection when not using box  (Windows)

jpleger, 2021-01-19:
## Summary:
I noticed that the process will crash when you select any tool other than the box selection... For example, when I select lasso or paint, the process will crash as soon as I click on the blue canvas.

## How to Reproduce: 

1. Open a schematic
2. Select anything but the box for selection, for example:

[image]

3. Start selection on schematic
4. (crash)

## Info
### Windows Version

[image]

### Error from app

[image]

### Log file (stderr)
```
[C:\VCS\sot23_breakout\horizon\sot23_breakout\top_sch.json] (horizon-eda:22572): GLib-GIO-WARNING **: 19:43:15.245: win32 session dbus binary not found: C:\Program Files\Horizon EDA\\gdbus.exe
[C:\VCS\sot23_breakout\horizon\sot23_breakout\top_sch.json] (horizon-eda:11132): GLib-GIO-WARNING **: 19:43:23.953: win32 session dbus binary not found: C:\Program Files\Horizon EDA\\gdbus.exe
```

Based on the logs, this looks like a similar issue that was in inkscape where gdbus wasn't being built: https://gitlab.com/inkscape/inkscape/-/issues/482

carrotIndustries, 2021-01-19:
This is very unlikely due to gdbus. Which Horizon version are you using?

jpleger, 2021-01-19:
Running the most current version, 1.3.0.

I can try removing, purging registry keys, etc. or run in debug mode, not sure the best way to troubleshoot since I am not familiar with the codebase :(

carrotIndustries, 2021-01-19:
Can you also try all other editors, i.e. package, frame, padstack etc. to see if the issue also occurs there. You can also try the latest build at https://ci.appveyor.com/project/carrotIndustries/horizon/builds/37326734/artifacts

jpleger, 2021-01-19:
I just tested and it does the same thing in the other editors.

```
[C:\VCS\electronic-parts\padstacks\padstack.json] (horizon-eda:24008): GLib-GIO-WARNING **: 16:22:26.617: win32 session dbus binary not found: C:\Program Files\Horizon EDA\\gdbus.exe
```

Using the latest build, it also appears to have the same behavior.

[image]

Log from stderr:

```
[C:\VCS\sot23_breakout\horizon\sot23_breakout\top_sch.json] (horizon-eda:8940): GLib-GIO-WARNING **: 16:25:53.275: win32 session dbus binary not found: C:\Users\jpleger\Downloads\horizon-2021-01-18-2252\horizon\\gdbus.exe
```

carrotIndustries, 2021-01-19:
Getting a stacktrace on windows isn't exactly easy as that'll involve building from source https://docs.horizon-eda.org/en/latest/build-win32.html and starting the editor with gdb attached. If you want, I can try and walk you through the process.

What's your GPU vendor? Can you reproduce this on another system?

jpleger, 2021-01-20:
Sorry, this is my Windows box so i'll need to setup a build environment, but that shouldn't be a big deal.

GPU is Nvidia 1080 card, I'll try to reproduce on another system here shortly.

jpleger, 2021-01-20:
Tested on another box with the most recent build...

Had slightly different behavior, on first try when I edited a default padstack it worked... then I closed, reopened and edited a padstack again and it popped up a different error.

[image]

Similar error about session dbus binary not found:

```
[C:\Users\jen.HOME\Documents\horizon-pool\packages\manufacturer\intel\EQFP-144\padstacks\pad.json] (horizon-eda:3224): GLib-GIO-WARNING **: 17:02:18.785: win32 session dbus binary not found: C:\Users\jen.HOME\Downloads\horizon-2021-01-18-2252\horizon\\gdbus.exe
```

Its a slightly different windows build:
[image]

carrotIndustries, 2021-01-20:
Good news, I managed to reproduce this on my windows test machine.

jpleger, 2021-01-20:
Awesome! I tried to get a build environment setup on windows last night, but was having difficulty getting it up and running (apologies, I'm not super familiar with msys2 environments).

I was looking at procmon and nothing immediately jumped out at me, but I'll admit that I am not very proficient in GTK on windows.

carrotIndustries, 2021-01-20:
Try https://github.com/horizon-eda/horizon/actions/runs/499863104 once the build has finished or compile the commit and confirm that this fixes the crash for you as well.

This only crashing on windows with nvidia drivers (didn't crash on intel) was pure chance, I'm glad you found it.

jpleger, 2021-01-22:
Confirmed this worked! Thanks so much, sorry I couldn't help more with finding the issue, not super familiar with the codebase :(

jpleger, 2021-01-22:
Sry, I didn't see if there was a PR that got merged, so will leave open.

commit 69ac4b50 (2021-01-22): CanvasGL line selection: fix potential memory corruption

the first render happend without any data being buffered

fixes #550
closed 2021-01-22 by carrotIndustries