horizon-eda/horizon #558: Installs the unstripped bin/horizon-eda executable

yurivict, 2021-01-31:
```
Warning: 'bin/horizon-eda' is not stripped consider trying INSTALL_TARGET=install-strip or using ${STRIP_CMD}
```

carrotIndustries, 2021-01-31:
Every linux distro I'm aware of (at least debian and arch) take care of stripping binaries during packaging, so I consider this to be a non-issue.

yurivict, 2021-01-31:
No, they have to do this by hand or not care about stripping at all. This is why this warning is there.

carrotIndustries, 2021-01-31:
> No, they have to do this by hand or not care about stripping at all

Arch Linux at least automatically strips binaries during packaging. Flatpak does so as well and extracts debug symbols while doing so.

In fact, `make install` installing non-stripped binaries is intended behaviour: https://www.gnu.org/prep/standards/html_node/Standard-Targets.html

> Do not strip executables when installing them.

Feel free to open a PR that adds the `install-strip` target if FreeBSD (guessing from context, should have been mentioned) really needs it.

yurivict, 2021-06-02:
> Arch Linux at least automatically strips binaries during packaging.

I don't think so. Arch just runs ```make install```.

> In fact, make install installing non-stripped binaries is intended behaviour: https://www.gnu.org/prep/standards/html_node/Standard-Targets.html

This document contains the paragraph:
```
‘install-strip’

    Like install, but strip the executable files while installing them. In simple cases, this target can use the install target in a simple way:

    install-strip:
            $(MAKE) INSTALL_PROGRAM='$(INSTALL_PROGRAM) -s' \
                    install
```
You don't have the ```install-strip``` target.

RX14, 2021-06-02:
> I don't think so. Arch just runs `make install`.

Arch linux's `makepkg` strips the binaries itself after `make` runs. This is what most linux distros do. I'm sure everyone would be happy to add an `install-strip` target, however it's not a bug that such a target doesn't exist, only a feature request.

carrotIndustries, 2021-06-02:
I'm really curious why FreeBSD needs such a target while any other package manager doesn't. How does FreeBSD handle the creation of debug symbol packages?

yurivict, 2021-06-02:
> I'm really curious why FreeBSD needs such a target while any other package manager doesn't. How does FreeBSD handle the creation of debug symbol package

This is because packages can be built with and without debug info. This is triggered from the port, and most GNU make projects strip binaries themselves because they have the ```install-strip``` target. Blanket stripping isn't good because it would be double-stripping in many cases.