horizon-eda/horizon #560: Updating PRs with just changing pin names in the unit file doesn't work

jue89, 2021-02-01:
The pool manager doesn't allow me to commit and push this change:
https://github.com/horizon-eda/horizon-pool/pull/161/commits/6597fe9288966cc1f26570c502f3e5a45bfd2e75
(I've added this commit manually.)

What I've done:
* Pick the PR in the Remote tab
* Click "Update PR"
* Go to the Unit tab, select the changed unit and click "Merge"
* Go back to the Remote tab and click "Update PR #..."

Nothing happens.

carrotIndustries, 2021-02-01:
That's odd. If got this correctly, 88ca98115bd32a99fc48cc4c812b44c75d3e6334 caused this for all PRs if they were cloned already. Wondering why it's only now that we notice this.

closed 2021-02-01 by carrotIndustries

jue89, 2021-02-02:
I can confirm that this commit fixes the issue.

commit 7ccfaaaf (2021-02-01): remote box: correctly set PR number

should fix #560