horizon-eda/horizon #564: Glib Exception: No GL implementation is available

tik, 2021-02-05:
Environment: VMware Workstation Pro 16, HostOS: Windows 10, GuestOS: Windows 10

When I open the Schematic or Board editor the following exception is thrown: "Glib::Error No GL implementation is available"

Ignoring the error, the editor windows open just fine and seem to work normaly (apart from a thick black border that is drawn around the window decorations).

Worstation Pro 16 supports OpenGL 4.1 and glView detects it ([opengl_details.txt](https://github.com/horizon-eda/horizon/files/5932956/opengl_details.txt)).

carrotIndustries, 2021-02-05:
> (apart from a thick black border that is drawn around the window decorations).

Can you share a screenshot of how this look like.

> the following exception is thrown: "Glib::Error No GL implementation is available"

I'm surprised that this doesn't completely crash the application.

I think that this is another opengl-on-windows related gtk bug. Can you install msys2 and run the opengl demo from `gtk3-demo` and post what this one outputs?

tik, 2021-02-06:
> 
> 
> > (apart from a thick black border that is drawn around the window decorations).
> 
> Can you share a screenshot of how this look like.

Sure, the borders loook like they could be the remains of the native window decorations: [schematic-editor](https://user-images.githubusercontent.com/77636/107110641-c4e25280-6849-11eb-876b-52aa15863b49.PNG)

> I'm surprised that this doesn't completely crash the application.
>
> I think that this is another opengl-on-windows related gtk bug. Can you install msys2 and run the opengl demo from `gtk3-demo` and [...]

Ok, this one is strange.

On the first run of the OpenGL demo it fails with the error message from the exception, translated to german. When I close the demo (not the gtk3-demo application, just the OpenGL demo window) and immediately open it again, it is working. The controls become visible once I click in the window:

[opengl-area_first-run](https://user-images.githubusercontent.com/77636/107110857-7e8df300-684b-11eb-811a-4f8617a514d2.PNG)
[opengl-area_second-run_start](https://user-images.githubusercontent.com/77636/107110928-155aaf80-684c-11eb-8e43-0b4dfadc6e1a.png)
[opengl-area_second-run_clicked](https://user-images.githubusercontent.com/77636/107110856-7df55c80-684b-11eb-8893-4b69d1eb2a94.PNG)

carrotIndustries, 2021-02-06:
> Sure, the borders look like they could be the remains of the native window decorations: schematic-editor

These should be semi-transparent shadows.

> Ok, this one is strange.

I try to fix bugs in depdencies whenever feasible and it affects a wide range of users. This issue however is related to an environment that probably very few people are going to use and I don't have a copy of VMware workstation.

I'm afraid, but the best advice I can give is to file bug with Gtk: https://gitlab.gnome.org/GNOME/gtk/issues/

tkornack, 2022-10-29:
My laptop (12th gen Intel integrated graphics) started throwing this error after upgrading to Ubuntu 22.10. I rebuilt the app from source (master). Every other GL app that I can find is working properly; the GTK demo app's GL Area works just fine. 

Other computers running 22.10 are working perfectly, so the problem might be specific to this newer hardware. Would be curious if there were any additional diagnostic information that might be helpful.

carrotIndustries, 2022-10-29:
> Ubuntu 22.10

Please open a new issue since this bug seems to be unrelated.

tkornack, 2022-10-29:
Turns out I hadn't properly removed the horizon-eda-upstream package. After that it all worked.