horizon-eda/horizon #565: Pool manager tab ordering and usability

MarSik, 2021-02-06:
Hi,

opening the Pool manager for the first time was a bit confusing as it displays the tabs in the following order:

Units Symbols Entities Padstacks Packages Parts Frames Settings

It seems the order follows the internal database ordering of objects (detailed -> generic). But if the preferred way to create the part is using the Part wizard it should probably be reversed to something like below as that show the Create part wizard button on the first tab users see.

Parts | Entities Symbols Units | Packages Padstacks | Frames | Settings

I would also consider adding the separator lines to show groups of objects that belong together.

carrotIndustries, 2021-02-10:
The separator's aren't supported by gtk, but I think that reordering the tabs makes sense as their current order is based on when they got added.

Any ideas where to put the parametric tabs?

MarSik, 2021-02-13:
Honestly, I have no idea atm. I need to try them first to see how they fit into the flow...