horizon-eda/horizon #566: Part import wizard - assign pins - show the package

MarSik, 2021-02-06:
The Assign pins screen of the Part import wizard requires me to enter pin names for each numbered pad. However I cannot double check which pad is which.

This is an issue any time the footprint is either not an industry standard or where vendors use different numbering in the datasheets. Think about BCE, ECB ordering of TO92 transistors or the same issue in SOT23, TO220 and other packages.

carrotIndustries, 2021-02-08:
How should the package preview show up? As an extra window that can be opened from the part wizard?

MarSik, 2021-02-09:
What about a side bar of the dialog (same window)? Would that work?

carrotIndustries, 2021-02-09:
This how it how it now looks like:
[image]

closed 2021-02-09 by carrotIndustries

MarSik, 2021-02-09:
This ia great! Thanks.

commit 18e12994 (2021-02-09): part wizard: show package preview

closes #566