horizon-eda/horizon #567: Pool manager - allow creating package directly from New part wizard

MarSik, 2021-02-07:
The first step when using the New part wizard is to select the package for the new part. In case the package does not exists yet, the user has to cancel the wizard, go to packages, create a new one, go back to the parts tab and start the wizard again. That is very inconvenient flow.

Can you add a way to start the new package editor from the wizard package selection dialog?

And while you are at it, what about exposing the Import KiCAD footprint there (and in the packages tab) too? I know the function is hidden in the Package editor, but that requires the user searches for it. A new button could easily serve as a shortcut - start package editor + import kicad footprint.

carrotIndustries, 2021-02-08:
> Can you add a way to start the new package editor from the wizard package selection dialog?

I don't really see why this is needed. The part wizard window isn't modal, so the user can create a package in the pool manager and then search for it in the part wizard's package selection page.

>  Import KiCAD footprint there (and in the packages tab

I think it makes sense to have in the packages tab, I'll see what I can come up with.

MarSik, 2021-02-09:
> I don't really see why this is needed. The part wizard window isn't modal

Hmm, that might work. However it is not obvious you can do that. And the only major thing I still remember about usability is that the rule of the game is "Do not make me think". But let me try what happens.

carrotIndustries, 2021-02-10:
> I think it makes sense to have in the packages tab, I'll see what I can come up with.

done in 8cbd93cefe1a006200e4ec53daf3305fcfbc2e57

closed 2021-02-10 by carrotIndustries