horizon-eda/horizon #568: Pool manager crashes when part from included pool is selected (and the other pool is disabled)

MarSik, 2021-02-07:
Reproducer using the Pool manager screen (the default screen after starting horizon-eda):

1) Add the horizon pool as usual
2) Add custom empty pool
3) Configure the custom pool and include the horizon pool into it
4) Disable horizon pool (or remove it completely)
5) Select custom pool
6) Click any unit, part or other entity..
7) Exception dialog appears..

After step 4 the pool link from custom pool to horizon pool still exists, but horizon-eda has no idea how to resolve the horizon pool UUID.

MarSik, 2021-02-07:
Btw, if I ignore the exception I can click the Edit button and I get reasonably good error message. Nothing that would point me to the fact that the issue is caused by a broken pool link though.

carrotIndustries, 2021-02-15:
The right way to fix this would be to force a pool update in that case.